siemens/ros-sharp #340: Unity 20

rboldu, 2020-08-23:
* [ ] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stack Overflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [ ] I have thoroughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

---
## I have a question! ##

* [ ] I searched the [Wiki](https://github.com/siemens/ros-sharp/wiki), [open](https://github.com/siemens/ros-sharp/issues) and [closed](https://github.com/siemens/ros-sharp/issues?q=is%3Aissue+is%3Aclosed) issues for an answer. I tried my best to find the answer by myself without success. I believe that the discussion we will have in this issue, and the solutions we might find, will help me, and likely other community members who have a similar problem.

**Here is my question:**

---
## I found a bug! ##
* [ ] I am using the latest ROS# version available here on the master branch.
* [ ] I am adding all required information, code and data files, screenshots and log files so that you can reproduce the problem.

* My OS is: 
* My Unity Version is: 
* My ROS Distribution is: 
* My Build Target Platform is: 

**Here is my bug description:**

Perform the following steps reproduce the bug:

Observed results:

Expected results:

Relevant Code:
```
<!-- Insert code here. Optionally attach code or data files to this issue. -->
```
---
## I have a feature request! ##

**I want to do this:**

**For being able to do it, I wish ROS# had the following feature:**

rboldu, 2020-08-23:
My OS is: Windows
My Unity Version is: 2020.1.0f1
My ROS Distribution is: Melodic

Bug description: 
I am facing errors creating a project and importing ROS# in Unity 2020.1.0f1. However, when creating a new project in the lower version e.g 2018.4.11f1 / 2019.4.8f1  it does work without any error.  
Is it compatible with 2020 version?

These are the errors:

"Assembly 'Library/ScriptAssemblies/Assembly-CSharp.dll' will not be loaded due to errors:
Reference has errors 'RosBridgeClient'."

"Assembly 'Library/ScriptAssemblies/Assembly-CSharp-Editor.dll' will not be loaded due to errors:
Reference has errors 'Assembly-CSharp'."

"Assembly 'Assets/RosSharp/Plugins/RosBridgeClient.dll' will not be loaded due to errors:
RosBridgeClient references strong named System.Threading.Tasks.Extensions in a different folder, versions has to match. Assembly references: 4.2.0.1 Found in project: 4.2.0.0."

closed 2020-08-29 by rboldu

repa1030, 2020-09-13:
I have the same Problem with 2020.1. How did you fix it?