siemens/ros-sharp #341: How to use dynamic reconfigure from ROS#

awatson3, 2020-09-02:
* [X] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stack Overflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [X] I have thoroughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

---
## I have a feature request! ##

**I want to do this:**
Use [dynamic_reconfigure](https://wiki.ros.org/dynamic_reconfigure) commands (such as param setting) from ROS#

**For being able to do it, I wish ROS# had the following feature:**
Please add an example tutorial here (https://github.com/siemens/ros-sharp/wiki) showing how to execute dynamic reconfigure commands with ROS#.  Many ROS API's (e.g. realsense, multisense, etc) require the use of dynamic_reconfigure to change settings at runtime and I'd like to do that using ROS#.

Context
> "dynamic_reconfigure and the Parameter Server (which getParam() uses) are designed to solve different problems. The parameter server is useful for situations where you know the value of parameters before a node starts, and know that those parameters are unlikely to change. dynamic_reconfigure is for situations where you might not know the values of parameters before a node starts, or those parameters might change while a node is running."
https://wiki.ros.org/ROS/Patterns/Parameterization#dynamic_reconfigure

berkayalpcakal, 2020-09-02:
Hello @awatson3,

I have not used dynamic_reconfigure package so far. But here are some recommendations from my side.

Looking at this [example](https://github.com/siemens/ros-sharp/blob/master/Libraries/RosBridgeClientTest/RosSocketConsoleExample.cs#L57), we can see that ROS# can call [rosapi services](https://github.com/RobotWebTools/rosbridge_suite/tree/develop/rosapi/srv) to get/set ros params, which then triggers the corresponding routines in [rosapi_node](https://github.com/RobotWebTools/rosbridge_suite/blob/develop/rosapi/scripts/rosapi_node).

Maybe you need to follow the similar steps to get/set dynamic_reconfigure parameters. What I see from the dynamic_reconfigure repository is that they provide a [rosservice](https://github.com/ros/dynamic_reconfigure/tree/60af0feb9dccd2dce5c01bc2256dd332fe2a862d/srv) - [server](https://github.com/ros/dynamic_reconfigure/blob/60af0feb9dccd2dce5c01bc2256dd332fe2a862d/src/dynamic_reconfigure/server.py), which you can call from Unity via ROS#.  

Another option would be implementing your own service server, which would run the bash script "rosrun dynamic_reconfigure dynparam COMMAND" (as described [here](https://wiki.ros.org/dynamic_reconfigure)) when the service is called.

One more option is to have equivalent ros params to dynamic_reconfigure parameters, which you can get/set using ROS# to update the ros params from Unity. Then again, using  your own custom service server, which would run the bash script "rosrun dynamic_reconfigure dynparam set_from_parameters" when the service is called.

I hope this helps. If you find a nice way of accomplishing what you want to do, please share your experiences here.

berkayalpcakal, 2020-10-07:
Hi again,

I successfully changed dynamic reconfigure parameters at runtime using ROS# by calling the ROS [service](https://github.com/ros/dynamic_reconfigure/blob/60af0feb9dccd2dce5c01bc2256dd332fe2a862d/src/dynamic_reconfigure/server.py#L83) advertised by the dynamic_reconfigure node. Here are the necessary steps;

* dynamic_reconfigure package provides [msg](https://github.com/ros/dynamic_reconfigure/tree/noetic-devel/msg) and [srv](https://github.com/ros/dynamic_reconfigure/tree/noetic-devel/srv) files. We need to create corresponding message classes and put them in our ROS# project. See [here](https://github.com/siemens/ros-sharp/wiki/Dev_NewMessageTypes#2-package-message) to create custom messages from a package.

* The following code snippet shows how to call the service from a console example. You just need to follow [this tutorials](http://wiki.ros.org/dynamic_reconfigure/Tutorials) to run dynamic_reconfigure server before running the code below.

 ``` cs
using System;
using RosSharp.RosBridgeClient;
using dynamic_reconfigure = RosSharp.RosBridgeClient.MessageTypes.DynamicReconfigure;

namespace RosSharp.RosBridgeClientTest
{
    public class RosSocketConsole
    {
        static readonly string uri = "ws://X.X.X.X:9090";

        public static void Main(string[] args)
        {
            RosSocket rosSocket = new RosSocket(new RosBridgeClient.Protocols.WebSocketNetProtocol(uri), RosSocket.SerializerEnum.Microsoft);

            dynamic_reconfigure.Config config_msg = new dynamic_reconfigure.Config()
            {
                bools   = new dynamic_reconfigure.BoolParameter[1]  { new dynamic_reconfigure.BoolParameter(name: "bool_param", value: true) },
                ints    = new dynamic_reconfigure.IntParameter[1]   { new dynamic_reconfigure.IntParameter(name: "int_param", value: 77) },
                strs    = new dynamic_reconfigure.StrParameter[1]   { new dynamic_reconfigure.StrParameter(name: "str_param", value: "dummy string") },
                doubles = new dynamic_reconfigure.DoubleParameter[1]{ new dynamic_reconfigure.DoubleParameter(name: "double_param", value: 0.33) }
            };

            rosSocket.CallService<dynamic_reconfigure.ReconfigureRequest, dynamic_reconfigure.ReconfigureResponse>
                ("/dynamic_tutorials/set_parameters", ServiceCallHandler, new dynamic_reconfigure.ReconfigureRequest { config = config_msg });

            Console.WriteLine("Press any key to close...");
            Console.ReadKey(true);
            rosSocket.Close();
        }

        private static void ServiceCallHandler(dynamic_reconfigure.ReconfigureResponse message)
        {
            Console.WriteLine("Config is updated. (e.g. double_param: " + message.config.doubles[0].value.ToString() + ")");
        }
    }
}
```

So this is how you can use ROS# to manipulate dynamic reconfigure params. You can have a similar script on Unity to accomplish the same task.

closed 2020-10-07 by berkayalpcakal