siemens/ros-sharp #346: ROS Bridge Connecting but not messages being received

cjonesy20, 2020-09-30:
* My OS is: Windows 10 for Unity, ROS Noetic is running on a Raspberry Pi 4 (Raspbian Buster)
* My Unity Version is: 2019.3.6f1
* My ROS Distribution is: Noetic
* My Build Target Platform is: Windows 10 initially then Android

Hi, I'm a TA at University of Illinois working on putting together a self driving car course. This is a small differential drive robot with an Ultrasonic sensor on a servo, Pi v2 RGB camera, BMO055 IMU, and two basic wheel encoders. ROSBridge is installed and running on port 9090. 

ROSConnector is in my Unity Heirarchy and the ROS Bridge URL is set. When I start the scene I see in Console Connected to RosBridge: ws:raspberrypi:9090 and I see the connection in ROS Bridge on the Raspberry Pi increment the Client connected totals. 

I used the generate messages functionality to create a custom message called Orientation. 

`using UnityEngine;

namespace RosSharp.RosBridgeClient
{
    public class IMUSubscriber : Subscriber<Messages.Orientation>
    {
        public float headingY;

        protected override void Start()
        {
            base.Start();
            Debug.Log("start");
        }

        protected override void ReceiveMessage(Messages.Orientation oMessage)
        {
            headingY = oMessage.heading_y;
            Debug.Log(headingY);
            Debug.Log(oMessage.header);
        }
    }
}`

I have added this IMUSubscriber to the scene and when I start the scene the "start" is printed to the console so I know at least the Start() method is being called. 

I can run rostopic echo /orientation and I see messages being sent through roscore so I know the rosnode for this topic is publishing. But the Debug.Log lines in ReceiveMessage never get run and I have no idea why it's not working. 

Brief screen capture walking through all this and showing how the data is not being received. https://www.youtube.com/watch?v=LTLqhbIxXow

berkayalpcakal, 2020-10-01:
Hello @cjonesy20,

There might be few possible reasons for not having a message flow even though a proper connection exists. 

First of all, as far as I understand from the video you have the old version of ROS#, please update it.

Once you have the current version of ROS# on your local, you will need to modify your message class definition a bit. Looking at the video around [1:02''](https://youtu.be/LTLqhbIxXow?t=62), it seems that the generated message does not follow the correct message class definitions given in [here](https://github.com/siemens/ros-sharp/wiki/Dev_NewMessageTypes#1-declare-a-new-message-type) (mind that the class attributes are c#-properties, and there is no JsonIgnore macro). You can take the ["geometry_msgs/Vector3"](https://github.com/siemens/ros-sharp/blob/master/Libraries/RosBridgeClient/MessageTypes/Geometry/msg/Vector3.cs) as an example, or you may want to use it directly?. 

If updating your ROS# does not help, you can refer to #309, #301. 
When you would like to deploy the app to the Android platform, you may find #287, #235 useful.   

I hope this helps.

cjonesy20, 2020-10-02:
I was using ROS# from the Unity Asset Store, previously. I created a new project and imported the one from this GIT Rep. I used the option to create all messages in a directory and it looks like the link you provided with no JsonIgnore. 

I'm not using URDF so #301 doesn't apply. For #309 I'm guessing I should just remove the RosConnector.cs from the scene and try that code directly? Does rosbridge need to be told about custom messages and what topics or does it expose all of ROS to any websocket connections to it?

EricVoll, 2020-10-02:
Are you building for UWP on Windows 10? If yes, then the IL2CPP compiler could make you the troubles. If you are building for UWP you can use [this](www.github.com/EricVoll/ros-sharp) fork.

berkayalpcakal, 2020-10-02:
@cjonesy20 

Please replace all the ROS# related scripts in your Unity project with the ones in this repository. unitypackage in the asset store might not be up to date. 

Once you have the current repo, I suggest that you try out the [RosSocketConsoleExample](https://github.com/siemens/ros-sharp/blob/master/Libraries/RosBridgeClientTest/RosSocketConsoleExample.cs) in debug mode so that you can observe which part of the code is not working, #309 would make much more sense then.  

I suspect that this is a problem with serialization problem. You can try out different combinations of protocol and serializer in the RosSocket constructor.

cjonesy20, 2020-10-02:
> Are you building for UWP on Windows 10? If yes, then the IL2CPP compiler could make you the troubles. If you are building for UWP you can use [this](www.github.com/EricVoll/ros-sharp) fork.
> Using the current (normal) version of Ros# on UWP platforms typically look like that. You can connect to the RosBridge but you're not receiving any data.

I'm not using UWP, just the standard Windows option. I also haven't tried a standalone build, just using the Editor right now. 

> @cjonesy20
>
> Please replace all the ROS# related scripts in your Unity project with the ones in this repository. unitypackage in the asset store might not be up to date.
>
> Once you have the current [...]

I created a new Unity project and used the code from this rep. No code from the asset store is being used. I tried the RosSocketConsoleExample.cs script. I can see it connect to ROSBridge on the Raspberry Pi side. I don't see anything print in the console except Press Any key to unsubscribe and Press any key to close. I put break points on all three methods and they never get called. I also tried WebSocketSharpProtocol and WebSocketNetProtocol, same result, both methods connect but no data shows up. 
Here's the output showing that the topics I wanted to access in Unity are being exposed by the ROS API. 
`pi@raspberrypi:~/ros_catkin_ws $ rosservice call /rosapi/topics
topics: [/rosout_agg, /rosout, /orientation, /client_count, /connected_clients, /encoder]
types: [rosgraph_msgs/Log, rosgraph_msgs/Log, picar/Orientation, std_msgs/Int32, rosbridge_msgs/ConnectedClients,
  picar/Encoder]
`

cjonesy20, 2020-10-03:
I was able to verify that rosbridge is working. I use the ROS bridge python client on the same PC as Unity and was able to connect to the same topic I'm trying to stream into Unity and it worked perfect.

berkayalpcakal, 2020-10-06:
Hi @cjonesy20,

[rosbridge protocol](https://github.com/RobotWebTools/rosbridge_suite/blob/groovy-devel/ROSBRIDGE_PROTOCOL.md) is used to communicate messages through rosbridge. Have you debugged the serialization of the messages while running the console example script? One should check the de-/serialized data whether the protocol requirements are satisfied or not.
But if you think that the issue is resolved, you can close it.

rassel, 2020-10-19:
I've been struggling a similar issue these days. 
-> [Play (Preview mode )] can subscribe topic, but [Build And Run] mode cannot subscribe topic.

My environment is below.
- Windows 10 64bit
- WSL(Ubuntu 16.04) kinetic
- Unity 2019 LTS
- ROS# v1.6

Today I managed to find a solution about this issue.
The solution is below.
It is important is to use The latest ROS#(final commit is 2020/8/11).
1. Set The latest  [Unity3D/Assets/RosSharp] under Unity Assets.
2. Build The latest  Libraries by Visual Studio 2019.
3. Replace  built "System.Threading.Tasks.Extensions.dll" with original RosSharp dll.

This solution enabled me to subscribe topics by both [Play] and [Build And Run] modes.

pegasus0642, 2020-12-17:
Hi, I have a similar issue. I use Eric's version for UWP and I need new message types as [visualization_msgs](http://wiki.ros.org/visualization_msgs) and I implemented all required message types under RosSharp.RosBridgeClient.MessageTypes.Visualization. 

[image]

When I connect to ROS it subscribes but I supposed to get NotImplementedException.

[image]

When I debug the code, I realised the ReceiveMessage(VisualizeTrajectoy message) never called. There is no error on both consoles. Do you have any idea?

- Windows 10
- Unity 2019.4.0
- ROS melodic
- Ubuntu 18.04

MartinBischoff, 2020-12-18:
Hi @pegasus0642
this looks like a serialization issue. Please check with @EricVoll which serializer to use for UWP. Also make sure that your visualization_msgs in .NET are correct. Any small typo causes the serializers to fail.

EricVoll, 2020-12-18:
Hi @pegasus0642 
thanks @MartinBischoff for notifying me.

While I can't see an issue with the `VisualizeTrajectory` class, I don't know how the `MarkerArray` class looks like. Could you also post that class in here? 
Also: Recently I generated a few MessageTypes using the Console Tools (which is just wonderful @MartinBischoff) I'd suggest that you put your msg file into the generator as well. Please note, that (currently) after generating you still have to change the `RosMessageName` property to the form you have currently, since I didn't adjust the ConsoleTool from the original repo.

A NotImplementedException sounds like some Reflection calls from the Json Deserializer can't find a method. To debug that: You can create a ConsoleApp in the Libaray project of the RosSharp library, and add your custom message type. Then subscribe to the topics and then you can step through the whole library to find the issue. 
In my version of Ros# all messages must implement a parameter less contructor. I can't see that on your VisualizeTrajectory class, but normally they exist implicitly if no other constructors exist.

pegasus0642, 2020-12-19:
Hi @EricVoll, thank you for your explanation. 

Here are my classes. I used [visualization_msgs/Marker](http://docs.ros.org/en/api/visualization_msgs/html/msg/Marker.html) and [visualization_msgs/MarkerArray](http://docs.ros.org/en/api/visualization_msgs/html/msg/MarkerArray.html). I don't know is there anything wrong here. You can see my MarkerArray class the below of Marker class.

Marker class:

```
using RosSharp.RosBridgeClient.MessageTypes.Std;
using RosSharp.RosBridgeClient.MessageTypes.Geometry;

namespace RosSharp.RosBridgeClient.MessageTypes.Visualization
{
    public class Marker : Message
    {
        public override string RosMessageName => "visualization_msgs/Marker";

        public const byte ARROW = 0;
        public const byte CUBE = 1;
        public const byte SPHERE = 2;
        public const byte CYLINDER = 3;
        public const byte LINE_STRIP = 4;
        public const byte LINE_LIST = 5;
        public const byte CUBE_LIST = 6;
        public const byte SPHERE_LIST = 7;
        public const byte POINTS = 8;
        public const byte TEXT_VIEW_FACING = 9;
        public const byte MESH_RESOURCE = 10;
        public const byte TRIANGLE_LIST = 11;
        public const byte ADD = 0;
        public const byte MODIFY = 1;
        public const byte DELETE = 2;
        public const byte DELETEALL = 3;

        public Marker()
        {
            this.Header = new Header();
            this.Ns = new String();
            this.Id = new Int32(); 
            this.Type = new Int32();
            this.Action = new Int32();
            this.Pose = new Pose();
            this.Scale = new Vector3();
            this.Color = new ColorRGBA();
            this.Lifetime = new Duration();
            this.Frame_locked = new Bool();
            this.Points = new Point[0];
            this.Colors = new ColorRGBA[0];
            this.Text = new String();
            this.Mesh_resource = new String();
            this.Mesh_use_embedded_materials = new Bool();
        }

        public Header Header { get; set; }
        public String Ns { get; set; }
        public Int32 Id { get; set; }
        public Int32 Type { get; set; }
        public Int32 Action { get; set; }
        public Pose Pose { get; set; }
        public Vector3 Scale { get; set; }
        public ColorRGBA Color { get; set; }
        public Duration Lifetime { get; set; }
        public Bool Frame_locked { get; set; }
        public Point[] Points { get; set; }
        public ColorRGBA[] Colors { get; set; }
        public String Text { get; set; }
        public String Mesh_resource { get; set; }
        public Bool Mesh_use_embedded_materials { get; set; }
    }
}

```
MarkerArray class:

```
namespace RosSharp.RosBridgeClient.MessageTypes.Visualization
{
    public class MarkerArray : Message
    {
        public override string RosMessageName => "visualization_msgs/MarkerArray";

        public Marker[] Markers { get; set; }
        public MarkerArray()
        {
            this.Markers = new Marker[0];
        }
    }
}
```

I also used the console app and still, I didn't get anything. No hits in message method during debugging.

[image]

EricVoll, 2020-12-27:
ok so I don't see an issue with the Message-Classes.
Since the console app doesn't use the Unity IL2CPP backend (which is usually the issue) the Code-Stripping is no problem as well.
Are you sure that the messages are published at all?
You can check that on the linux machine using `rostopic echo /visualize_trajectory_curve` in the terminal.
If the messages are published successfully, then there might be issues with some names etc. such that the deserializer doesn't find the right subscriber. In the Console-App you can step into the RosConnector WebSocket where the deserialization happens. This would be here: https://github.com/EricVoll/ros-sharp/blob/e6fb59d05f5886b774311e97cadf77b81e305bca/Libraries/RosBridgeClient/RosSocket.cs#L192 You can set a break point there and check if the messages reach your console app at all. (Hint: Use a conditional break point to filter out the message types you are looking for)

If the messages reaches the RosConnector, then there is an issue with the deserialization. Otherwise you have a problem with publishing the messages. 
Another issue might be that the subscribing fails. Could you show me the console output of both the linux and Console app side?

ZahraaBass, 2021-01-20:
Hello, I'm facing a similar problem but with LaserScan msgs.

In my case I was able to receive fake odometry data from ROS through a talker node publishing on the /chatter_ topic. A link showing what I did: https://www.youtube.com/watch?v=yXWNSwk4Irs
I was also able to import the turtlebot URDF File: https://www.youtube.com/watch?v=jSc0PF_tSAQ

However, when I use the LaserScanSubscriber, and subscribe to the /scan topic, the DebugLog print in ReceiveMessage are not printing on the Console. Basically what I'm doing is launching the husky gazebo simulation, which publishes Lidar Data on the /scan topic like ranges. On the ROS side, the /scan topic shows one publisher (gazebo) and one subscriber (Rosbridge). so the problem is on the Unity side.. I've been trying to solve this problem for more than a week and I'm desperate for some clue. I'm kinda a newbie to both (ROS and Unity) and would really appreciate some help/insight.

I also made a video documenting what my problem is: https://www.youtube.com/watch?v=L9aU0XW6G1o
Should I open a new issue for this?

Ragingram2, 2021-04-23:
Hey everyone I am having the same problem but with all of the subscribers. 

- Windows 10
- Unity 2020.3.1f
- ROS2 Foxy Fitzroy
- Ubuntu 20.04 (Windows Subsystem Linux)

This is the output from the console. I am running a C++ publishing node and the ros2 version of ROS_Bridge. I print out every message sent and received by ROS_Bridge.
[image]

here is my unity project. 
[image]
The image subscriber is mostly unchanged (I added a method so that I can print from a dll(Yes I know, very hacky)).

As you can see Unity says it connects to ROS_Bridge, and ROS_Bridge receives the message that lets it know that Unity is subscribing to the "/image" topic. ROS_Bridge then starts to send the messages but Unity is not receiveing them. I tried debugging all the Unity Ros-sharp code and was led to the ROS-sharp dll. I downloaded the source code and after implementing a hacky console debugging solution, I saw that the websocket wasn't getting a any messages at all (Both WebSocketSharp and WebSocketNet). 

My gut tells me this probably has something to do with me using WSL for Linux, although from what I've read it should be completely fine. 
Another possible failure point is ROS_Bridge. The branch I am using is being developed for ROS2 but is severly WIP. I had to fix many typos, and change a few lines just to get it working. Although it when I publish from Unity rosbridge sees them and prints out the message. So the problem is purely with subscribing in Unity.

If any one has experience doing this with WSL or the ros2 branch of ROS_bridge, any help would be very much appreciated. If you need any more details please let me know. Thanks!

MartinBischoff, 2021-04-26:
Hi @Ragingram2 ,
did you inform the ROS_Bridge team about your fixes. Maybe they can support you getting the communication running on your platforms.

Ragingram2, 2021-04-26:
Hey @MartinBischoff,
I have not yet communicated with them. I was hoping to do so once my implementation was more or less complete, so that I could create a PR with as many fixes as I could find. But from your reply it sounds like you think the issue might be in the Ros_bridge side. 

If anyone else has a different solution, that would also be appreciated.

Thank you for the help, I'm now gonna go and compile all of my fixes into a somewhat readable form lol. 

P.S: this is a really great library, and probably saved our team lots of dev time. Thanks!

closed 2021-08-10 by MartinBischoff