siemens/ros-sharp #347: Change ROS Bridge Server URL in code?

fredrikbaberg, 2020-10-07:
* [x] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stack Overflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [x] I have thoroughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

---
## I have a question! ##

* [x] I searched the [Wiki](https://github.com/siemens/ros-sharp/wiki), [open](https://github.com/siemens/ros-sharp/issues) and [closed](https://github.com/siemens/ros-sharp/issues?q=is%3Aissue+is%3Aclosed) issues for an answer. I tried my best to find the answer by myself without success. I believe that the discussion we will have in this issue, and the solutions we might find, will help me, and likely other community members who have a similar problem.

**Here is my question:**

Is it possible to change the ROS Bridge Server URL through code?

My `roscore` runs on different devices, which means I sometimes need to specify a different URL. To do this I need to modify the URL of the `RosComponent` and build again.

I've created a menu with a script that modifies `RosBridgeServerURL` of the `RosConnector` component, however, I can't find a way for it to use the updated URL.
I've considered using a script to replace the `RosConnector` in runtime, but as other ROS# depends on it this was not as straightforward as I had hoped.

I am not sure if this should be a feature request, so I've started with a question.

berkayalpcakal, 2020-10-07:
> My `roscore` runs on different devices, which means I sometimes need to specify a different URL. To do this I need to modify the URL of the `RosComponent` and build again.

I did not quite get the motivation behind such request. You can always make different devices communicate with each other under one 'ROS master', which allows you to access all the data flow over a single device. See more about this [here](http://wiki.ros.org/ROS/Tutorials/MultipleMachines).
 
If you insist on keeping your ROS environments separate from each other, you can think about having config files like "url_device1.txt", "url_device2.txt" .., so that you can instantiate different RosSocket's in Unity by parsing the config files at run time. See [Text Assets](https://docs.unity3d.com/Manual/class-TextAsset.html)

I hope this helps.

fredrikbaberg, 2020-10-07:
Thank you for the reply.

> I did not quite get the motivation behind such request. You can always make different devices communicate with each other under one 'ROS master', which allows you to access all the data flow over a single device. See more about this [here](http://wiki.ros.org/ROS/Tutorials/MultipleMachines).

I'll try to clarify my use case a bit, I see now that I did not formulate it clearly. There is a single 'ROS master' but I cannot assume the URL is known when I build the Unity3D project.

I'm doing all development on a single laptop which can run simple simulations. As I'm now increasing complexity I need to offload some things to another machine, in this case either Unity3D (previously Gazebo) or ROS.
As I'm not always on the same network, the IP for the ROS Bridge Server will change. I know the IP when I launch, but not when I build.

> If you insist on keeping your ROS environments separate from each other, you can think about having config files like "url_device1.txt", "url_device2.txt" .., so that you can instantiate different RosSocket's in Unity by parsing the config files at run time. See [Text Assets](https://docs.unity3d.com/Manual/class-TextAsset.html). (Or some other fancy config file readers/writers that you can search on Unity forums.)

In this case I'd prefer using a UI Text element, but the idea should be similar. I think the issue becomes: How/where do I instantiate a Ros Socket _at run time_?

As far as I can tell this is done in RosConnector.cs Awake(), where it tries to connect directly. If I understood Awake correctly it is called first, before Start, but as the execution order is non-deterministic between all Awake() calls in the project I cannot guarantee to be able to modify it in time.
Does this leave me with modifying RosConnector.cs so that I can pass a URL and ask it to connect again? Or is there another option which I have missed?

MartinBischoff, 2020-10-07:
Hi @fredrikbaberg ,
thanks for explaning your use case in more detail. Yes, you will have to modify `RosConnector.cs`.
Here is a previous pull request which could serve as a reference for the modifications you require: https://github.com/siemens/ros-sharp/pull/231
The code there makes RosConnector to automatically reconnect on disconnect at runtime. Changing the URL remains to be done.
Best, Martin

fredrikbaberg, 2020-10-07:
Thank you @MartinBischoff, that seems very useful. I had hoped not having to modify the core files, but now I know it is needed.

I spent some more time considering the suggestion @berkayalpcakal made about using text files and ended up with a workaround with minimal changes to RosConnector.cs, where I (re)load the scene instead.
With [PlayerPrefs](https://docs.unity3d.com/ScriptReference/PlayerPrefs.html) I had to add one line to RosConnector.cs, in Awake():
> RosBridgeServerUrl = PlayerPrefs.GetString("RosBridgeServerURL");

I collect the URL with user input, store it using PlayerPrefs, and reload the scene. For now that should be enough.

I'll close this issue since it has been answered.

closed 2020-10-07 by fredrikbaberg