siemens/ros-sharp #348: ROS Bridge connects but throws an error

yohannes-taye, 2020-10-16:
## I have a question! ##

**Here is my question:**
I have Ubuntu 14 setup on a VM and Unity 2019.4 on Windows. I have cloned the repo and started following the tutorials. Everything went well with setting up ROS, Turtle bot, and Gazebo. It worked till I reached the part of the tutorial 2.1 Transfer URDF from ROS to Unity. On the Unity side this is what I get 
[image]

While on the ros side it shows the following error 
[image]

At this point im stuck and dont know what else to try. I have looked at several of the issues on this repo and the issue still persists.

berkayalpcakal, 2020-10-16:
Hello @yohannes-taye ,

I have searched this issue on web, it seems like this issue is related to your ROS environment on your Ubuntu. Here are some suggestions that I found;

* `sudo apt-get update && sudo apt-get upgrade` 
* `sudo apt install ros-$ROS_DISTRO-rosbridge-suite`
* `sudo apt install ros-$ROS_DISTRO-genpy `

I hope this helps.

yohannes-taye, 2020-10-19:
Thank you so much @berkayalpcakal!! That fixed my issue :)

closed 2020-10-19 by yohannes-taye