siemens/ros-sharp #350: Fibonacci action server bug

m1541, 2020-10-19:
## I found a bug! ##
* [x] I am using the latest ROS# version available here on the master branch.
* [x] I am adding all required information, code and data files, screenshots and log files so that you can reproduce the problem.

* My OS is: Ubuntu 18.04
* My Unity Version is: 2019.3.0f1
* My ROS Distribution is: Melodic
* My Build Target Platform is: Linux

**Here is my bug description:**
Hi, I'm having a problem implementing an action server in rossharp. I've tried with the basic Fibonacci server and everything seems fine until the result is sent back to the client. Has anyone encountered this bug before?

Perform the following steps reproduce the bug:
Follow the Fibonacci action server tutorial -> https://github.com/siemens/ros-sharp/wiki/User_App_Fibonacci_Action_Server

Observed results:
[image]

Expected results:
Action server returning result to the client.

berkayalpcakal, 2020-11-03:
Hello @m1541,

I could reproduced this error only when I run the `rosrun actionlib_tutorials fibonacci_client`. 
When I trigger the server by directly publishing to `/fibonacci/goal` topic, no such problem occurs.

The fibonacci server on the Unity side never publishes "LOST" goal status, you can verify this via  `rostopic echo /fibonacci/status | grep 'status: 9'`. So I am not quite sure how the client could end up with the LOST status.
Once the action server on the Unity side finishes the action, it publishes the goal status "SUCCEEDED" as we can also see it from the screenshot you provided.

berkayalpcakal, 2020-11-03:
One interesting observation I just had;

If I run the fibonacci client via `rosrun actionlib_tutorials fibonacci_client`, I can reproduce the error.
But if I run it via `rosrun actionlib_tutorials fibonacci_client.py`, then no such error occurs. 

I will close this issue since the issue can be related to implementation of the client script or its executable. I can set/cancel goals by interacting with the server topics without any problem.
If someone faces a problem with the ROS# implementation of the server/client on the Unity side, the issue can be opened again.

closed 2020-11-03 by berkayalpcakal

danielecavestri, 2021-06-08:
Hi, I'm having the same problem as @m1541.

- My OS is: Windows 10 as host, Ubuntu 18.04 on VMware 16 Player as guest (network adapter is bridged)
- My Unity Version is: 2019.4.26f1
- My ROS Distribution is: Melodic
- My Build Target Platform is: Windows

In particular I am using the [smach ](http://wiki.ros.org/smach) package of ROS to run a state machine in which evrey state can call an action. Then the state machine proceeds basing on the result received from an action server. I have the action server running on Unity.

The above mentioned problem cause every state of the state machine to finish as `aborted`. The expected behaviour is `succeded`.  To reproduce this error I created a simple state machine which calls FibonacciActions, here is the code:

```py
#!/usr/bin/env python

import rospy
import smach
import smach_ros
from actionlib_tutorials.msg import FibonacciAction, FibonacciGoal
from smach_ros import SimpleActionState

if __name__ == '__main__':
    rospy.init_node('smach_action_state_machine')
    sm = smach.StateMachine(['sm_succeeded', 'sm_aborted', 'sm_preempted'])
    with sm:
        sm.add('first_task',
               SimpleActionState('fibonacci',
                                 FibonacciAction,
                                 goal=FibonacciGoal(order=5)),
               transitions={'succeeded': 'second_task', 'preempted': 'sm_preempted', 'aborted': 'sm_aborted'})

        sm.add('second_task',
               SimpleActionState('fibonacci',
                                 FibonacciAction,
                                 goal=FibonacciGoal(order=10)),
               transitions={'succeeded': 'sm_succeeded', 'preempted': 'sm_preempted', 'aborted': 'sm_aborted'})

    sis = smach_ros.IntrospectionServer('my_server_name', sm, '/MY_SM_ROOT')
    sis.start()

    outcome = sm.execute()
    rospy.spin()
    sis.stop()
```

As @berkayalpcakal mentioned, running `rosrun actionlib_tutorials fibonacci_client.py` does not cause problems. But I think the problem can be on the ROS# Unity side because, running my state machine and `rosrun actionlib_tutorials fibonacci_server` as action server, results in the following expected behaviour:
[image]

On the other hand, running the state machine and `UnityFibonacciActionServer.cs` as action server, results in the following wrong behaviour:
[image]
[image]

## Possible cause: Action Server difference
I have noticed that when running `UnityFibonacciActionServer.cs` on Unity and `rosrun actionlib_tutorials fibonacci_client` on ROS no goal_id is provided (run `rostopic echo /fibonacci/status` to check):
[image]

While running `rosrun actionlib_tutorials fibonacci_server` and `rosrun actionlib_tutorials fibonacci_client` a goal_id is provided:
[image]

I don't know if this missing goal_id can cause the problem, but it is for sure a difference caused by the implementation of `UnityFibonacciActionServer.cs` (assuming that the correct behaviour is to have a non-null goal_id)