siemens/ros-sharp #351: Question about correct use of LookupTransformActions

EricVoll, 2020-10-20:
* [x] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stack Overflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [x] I have thoroughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

---
## I have a question! ##

* [x] I searched the [Wiki](https://github.com/siemens/ros-sharp/wiki), [open](https://github.com/siemens/ros-sharp/issues) and [closed](https://github.com/siemens/ros-sharp/issues?q=is%3Aissue+is%3Aclosed) issues for an answer. I tried my best to find the answer by myself without success. I believe that the discussion we will have in this issue, and the solutions we might find, will help me, and likely other community members who have a similar problem.

**Here is my question:**

What is the suggested way to do `TransfromLookup`s?

I tried to read through the `FibonacciActionClient` example and tried to follow the concept behind it. But when trying to create a similar class for the TransfromLookup I noticed that the `TransfromLookup {Goal, Feedback, Result ...}` all inherit from `Message` and not as in the case for `FibonnaciActionResult` etc. (Example: `TransfromLookupActionResult` inherits from `Message`, but `FibonacciActionResult` inherits from `Action<FibonacciResult>`) This means I'd have to re-write all the TransfromLookup classes, which seems odd.

So I guess my question is: 
- Did I understand the way how to use TransfromLookups wrong?
- If not, and if one has to implement a similar thing as the FibonacciActionClient: Why include TransfromLookup classes, which adhere to the same naming convention as the Fibonacci-Classes, but have a completely different structure?

```

berkayalpcakal, 2020-10-22:
Hi @EricVoll,

Sorry but i did not get what you mean by "doing TransformLookup"? Are you trying to run an action client application with your custom action definition? You should first have an ".action" file, please see [here](http://wiki.ros.org/actionlib). Once you have your action definition you can see on your ROS environment that there are 7 messages automatically generated by ROS. Then you can generate the equivalent C# message classes as described [here](https://github.com/siemens/ros-sharp/wiki/Dev_NewMessageTypes).

I suggest that you first go through the ROS tutorials, so that you would have an understanding of the inheritance pattern among ROS-generated 7 action messages.

EricVoll, 2020-10-26:
Hi @berkayalpcakal 

Yes. As you said, I'm trying to run an action client. But I guess the issue might not be Ros# but a lacking understanding of the underlying ros concept 🙂
I'll dig into the ROS tutorials first.

Thanks for the reply!

closed 2020-10-26 by EricVoll