siemens/ros-sharp #352: [ROS2] Image Publisher Problems

maxlein, 2020-10-27:
* [x] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stack Overflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [x] I have thoroughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

---
## I have a question! ##

* [x] I searched the [Wiki](https://github.com/siemens/ros-sharp/wiki), [open](https://github.com/siemens/ros-sharp/issues) and [closed](https://github.com/siemens/ros-sharp/issues?q=is%3Aissue+is%3Aclosed) issues for an answer. I tried my best to find the answer by myself without success. I believe that the discussion we will have in this issue, and the solutions we might find, will help me, and likely other community members who have a similar problem.

**Here is my question:**

I am trying to get a video stream working from unity to ros foxy.  
Therefore I tried to use the [ImagePublisher](https://github.com/siemens/ros-sharp/blob/master/Unity3D/Assets/RosSharp/Scripts/RosBridgeClient/RosCommuncation/ImagePublisher.cs) but the camera render callback ( `Camera.onPostRender += UpdateImage;` ) was never triggered.  

So I adapted it to use the RenderPipelineManager and also made a version for Compressed and Raw Images.  
Quick and dirty to get things running:  

<details>
  <summary>CompressedImagePublisher</summary>  

```cs
/*
© CentraleSupelec, 2017
Author: Dr. Jeremy Fix (jeremy.fix@centralesupelec.fr)

Licensed under the Apache License, Version 2.0 (the "License");
you may not use this file except in compliance with the License.
You may obtain a copy of the License at
<http://www.apache.org/licenses/LICENSE-2.0>.
Unless required by applicable law or agreed to in writing, software
distributed under the License is distributed on an "AS IS" BASIS,
WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
See the License for the specific language governing permissions and
limitations under the License.
*/

// Adjustments to new Publication Timing and Execution Framework 
// © Siemens AG, 2018, Dr. Martin Bischoff (martin.bischoff@siemens.com)

using System;
using UnityEngine;
using UnityEngine.Rendering;

namespace RosSharp.RosBridgeClient
{
    public class CompressedImagePublisher : UnityPublisher<MessageTypes.Sensor.CompressedImage>
    {
        public Camera ImageCamera;
        public string FrameId = "Camera";
        public int resolutionWidth = 640;
        public int resolutionHeight = 480;
        [Range(0, 100)]
        public int qualityLevel = 50;
        public float scanPeriod = 0.4f;
        
        private MessageTypes.Sensor.CompressedImage message;
        private Texture2D texture2D;
        private Rect rect;
        private float previousScanTime = 0;

        protected override void Start()
        {
            base.Start();
            InitializeGameObject();
            InitializeMessage();
        }

        void OnEnable()
        {
            RenderPipelineManager.endCameraRendering += RenderPipelineManager_endCameraRendering;
        }
[... 188 lines cut ...]
    nanosec: 536071062
  frame_id: base_link
height: 480
width: 640
encoding: rgba8
is_bigendian: 0
step: 0
data:
- 0
- 65  
...
```

So my question is if there are other guys here who are testing this with ros2?  
I am a bit clueless atm.  

The compressed image publisher has the same configuration but shows nothing.  
And the raw publisher seems to have a wrong encoding but finding the correct one between unity and ros is a bit confusing.  

So again, maybe here are some guys with more experience than me ;)

* My OS is: Ubuntu 20.04
* My Unity Version is: 2020.1.7f1
* My ROS Distribution is: foxy
* My Build Target Platform is: x86

jim18324, 2021-04-21:
@maxlein Have you solved this problem?

maxlein, 2021-04-24:
No

MartinBischoff, 2021-04-26:
Hi @jim18324 and @maxlein , 
I had the same problem when writing the ImageSubscriber and Publisher for ROS. Info about the image encoding on both sides ROS and Unity can hardly be found.
In ROS there was an additional (redundant) header in the beginning of the image byte array that caused me a lot of trouble. [See my old post on stackoverflow.](https://stackoverflow.com/questions/42231191/reading-a-ros-compressedimage-data-byte-array-in-c-sharp/42248628#42248628)
* Maybe the header bytes have been removed in ROS2?
* You can test image communication between ROS and ROS2 to see if the codec was changed. In any case opencv should be able to load the image.
* Loading the image contents in an hex editor helped me finding the answer.

I hope these hints help; sorry I can't support more.

jim18324, 2021-04-27:
@MartinBischoff thank you for your hints. 
Forgive me I am new to Unity and ROS. I have a stupid question, how does the ROS# deal with this additional header problem when sending an image to ROS?

MartinBischoff, 2021-04-27:
@jim18324 after reading my old posts more carefully I remember that the header 12byte header exisited in the Compressed**Depth**Image only, not in the CompressedImage. I needed that for the  [3d point cloud visualization](https://www.youtube.com/watch?v=OytzagQirrk) only.
In ROS# we just have CompressedImage subscribers and publishers which don't have to deal with the 12 bytes in beginning.
Sorry for the confusion; maybe the hints still help you to track down the problem.

jim18324, 2021-04-27:
@MartinBischoff thank you for your fast reply. I was wondering is there somewhere in the Ros# that might edit or modify the image data? Because, I've check the length of the image data passing into ros2-web-bridge and its length increase! (I also ask this question at #389 )

MartinBischoff, 2021-04-27:
answered in #389 .

closed 2021-08-10 by MartinBischoff

FlorGrosso, 2021-09-03:
@maxlein @jim18324 were you able to solve this issue? I'm having it as well and I'm unable to get proper images in ROS 2. Any hint you may have will be helpful, thanks!

jim18324, 2021-09-15:
@FlorGrosso Sorry for late reply, but i haven't solve it, i think the problem is serialize and deserialize the data.