siemens/ros-sharp #355: Slow Image Publishing

jpapon, 2020-11-06:
* [X] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stack Overflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [X] I have thoroughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

---
## I have a question! ##

* [X ] I searched the [Wiki](https://github.com/siemens/ros-sharp/wiki), [open](https://github.com/siemens/ros-sharp/issues) and [closed](https://github.com/siemens/ros-sharp/issues?q=is%3Aissue+is%3Aclosed) issues for an answer. I tried my best to find the answer by myself without success. I believe that the discussion we will have in this issue, and the solutions we might find, will help me, and likely other community members who have a similar problem.

**Here is my question:**

I've been using ROS-sharp to publish out images from Unity3D into ROS.
Unfortunately, the framerates I achieve are very low. 
I tried changing to raw images instead of compressed, but this made the problem worse.
Did some timing, and the jpeg compression was fast for my images (~10ms), so looked for what was causing the delay.
I've tracked down the main culprit to the `WebSocket.SendAsync(data, null);` 

This appears to be the same problem in this issue: https://github.com/sta/websocket-sharp/issues/471

What can I do to increase framerates for images going into ROS? As is, for 1280x720 images I can't get more than ~15fps.
Have you experimented with any ways to speed this up?

Thank you!

berkayalpcakal, 2020-11-07:
Hi @jpapon,

I did few experiments myself, and here I my observations and thoughts on this issue.

In order to make a fair judgement on the performance of the websocket communication, I first checked the frame rate of the Unity application itself, then I checked the frequency of received ROS messages via 'rostopic hz /image_topic'.

So, on my pc the Unity application runs at ~29 fps when I have an additional camera of resolution 1280x720 in the scene, of which the frames are then published as compressed images using the default ImagePublisher of the repository. On the ROS side, the frequency of the received messages are observed as ~24 fps.  
When the resolution of the camera is 1920x1080, the numbers are ~14fps and ~13 fps respectively. 

In the aforementioned issue, it is said that the application runs at 30 Hz, yet the the messages are published at 5 Hz. 
Are you sure that your Unity application itself is running at a relatively higher fps, and yet msg publishing is only at ~15fps? 
The aforementioned issue refers to the data of size ~320Kb, whereas the compressed image of 1280x720 is encoded to a msg of size ~20Kb in my case. I think the biggest bottleneck is rather the rendering of the camera as far as I can observe from the Unity profiler tool. On my pc the camera rendering takes ~21 ms, the compression takes ~10 ms.

berkayalpcakal, 2020-11-07:
I experimented one more setup. Instead of rendering/publishing the 1280x720 images at every Unity frame, I did split the images into two chunks of size 640x720, and rendered/published each of the chunks alternately at one Unity frame. So the Unity application framerate increased to ~ 60 fps, and the rate of receiving messages on the ROS side increased to ~48 fps. 

The aim was to reduce the amount of data to be transmitted, and see its effect on the websocket communication speed.
The observation that the framerates are halved when the data size is doubled, makes me guess that websocket communication does not create an additional communication bottleneck with such small data sizes.  

_PS: I simply changed the [rect, and the texture2D](https://github.com/siemens/ros-sharp/blob/master/Unity3D/Assets/RosSharp/Scripts/RosBridgeClient/RosCommuncation/ImagePublisher.cs#L53) definitions dynamically. You can split an image to be rendered/published into any amount of chunks, which are to be merged back to a single image on the receiver side with an additional cost of course._

jpapon, 2020-11-10:
Hey @berkayalpcakal how do you have your websocket configured?

In my experiments I was just using an empty scene, so the rendering time was negligible. If I turn off the image publisher, the scene runs at 800fps. I timed the compression call in the script, and found it to be ~10 ms for 1280x720, and ~20ms for 1920x1080. I also modified the script to publish uncompressed images as well to eliminate compression time entirely. 
In all cases, there was an additional substantial chunk of time spent on the CPU which limits the framerate - as far as I can tell, this was the websocket call.

MartinBischoff, 2024-01-22:
outdated / closing due to inactivity

closed 2024-01-22 by MartinBischoff