siemens/ros-sharp #358: Service server (in Unity) doesn't receive the request from ROS client

Rive4, 2020-11-19:
* [x] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stack Overflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [x] I have thoroughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

---
## I have a question! ##

* [x] I searched the [Wiki](https://github.com/siemens/ros-sharp/wiki), [open](https://github.com/siemens/ros-sharp/issues) and [closed](https://github.com/siemens/ros-sharp/issues?q=is%3Aissue+is%3Aclosed) issues for an answer. I tried my best to find the answer by myself without success. I believe that the discussion we will have in this issue, and the solutions we might find, will help me, and likely other community members who have a similar problem.

**Here is my question:**

Hello,

First of all I would like to thank you for the time you spent woking in this package, it is really interesting and useful.
I post as a question because I'm not sure if it is my problem or a bug.

I am trying to create a service in Unity, specifically GetModelState service (the same one as gazebo has).
The problem is that when I call the service from ROS using python script or through console, the call gets stuck and never returns a message.
I created GetModelStateRequest and GetModelStateResponse from Message as you can see below:
```
public class GetModelStateRequest : Message
    {
        [JsonIgnore]
        public const string RosMessageName = "elkarbot_sim/GetModelState";

        public RosSharp.RosBridgeClient.MessageTypes.Std.String model_name;
        public RosSharp.RosBridgeClient.MessageTypes.Std.String relative_entity_name;

        public GetModelStateRequest(RosSharp.RosBridgeClient.MessageTypes.Std.String _model_name, RosSharp.RosBridgeClient.MessageTypes.Std.String _relative_entity_name)
        {
            model_name = _model_name;
            relative_entity_name = _relative_entity_name;
        }
    }

    public class GetModelStateResponse : Message
    {
        [JsonIgnore]
        public const string RosMessageName = "elkarbot_sim/GetModelState";

        public RosSharp.RosBridgeClient.MessageTypes.Std.Header header;
        public RosSharp.RosBridgeClient.MessageTypes.Geometry.Pose pose;
        public RosSharp.RosBridgeClient.MessageTypes.Geometry.Twist twist;
        public RosSharp.RosBridgeClient.MessageTypes.Std.Bool success;
        public RosSharp.RosBridgeClient.MessageTypes.Std.String status_message;

        public GetModelStateResponse(){
        }

    }
```
(Note: I know that it is possible to use 'using', I only wanted to be sure that the problem was not that.)

I also overrided ServiceCallHandler function which is necessary to handle with the request. The code starts like this:

```
        protected override bool ServiceCallHandler(GetModelStateRequest request, out GetModelStateResponse response)
        {
            print("Service has been called.");
            ...
            ...
       }
```

A strange behaviour that I found in this problem is that when I called the service through the console using autocomplete with null values, the message was received in Unity, but when I filled the parameters with strings, the ServiceCallHandler was never executed.

**Things that I took into account before post this problem:**
1) I made catkin_make with all the messages and services defined, CMakeList updated and sourced the workspace before execute rosbridge websocket and the service call.
2) The service is listed in ROS (executing 'rosservice list').
3) The call is sent by the client. (I checked this using wireshark).

I am working in Ubuntu 18.04, Unity 2019.3.13f1 for Ubuntu and ROS Melodic.

Thank you,
Andoni

Rive4, 2020-11-19:
Well, seem that I found what was my problem.
**I was using RosBridgeClient.MessageTypes.Std.String instead of UnityEngine.String type.**
I am not sure if RosBridgeClient.MessageTypes.Std.String is used in the rest of ROS# but seem to be quite confusing that ithere are available RosBridgeClient.MessageTypes namespace classes and but to generate C# classes for services, you shouldn't use them (e.g. strings or booleans), isn't it?

Hope this helps someone if he/she makes my same mistake :)

MartinBischoff, 2020-11-19:
Hi @Rive4 !

Great that you found the error by yourself.

Here is the difference:
[RosBridgeClient.MessageTypes.Std.String](https://github.com/siemens/ros-sharp/blob/master/Libraries/RosBridgeClient/MessageTypes/Std/msg/String.cs) corresponds to the [ROS message String](http://docs.ros.org/en/melodic/api/std_msgs/html/msg/String.html).
It contains data of type  [UnityEngine.String](https://docs.unity3d.com/ScriptReference/String.html) or just [string](https://docs.microsoft.com/en-us/dotnet/api/system.string?view=net-5.0). 

Best,
Martin

closed 2020-11-19 by MartinBischoff