siemens/ros-sharp #362: URDF Plugin script is able to upload gazebo plugins?

Tonirayn, 2020-12-01:
* [x] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stack Overflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [x] I have thoroughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

---
## I have a question! ##

* [X] I searched the [Wiki](https://github.com/siemens/ros-sharp/wiki), [open](https://github.com/siemens/ros-sharp/issues) and [closed](https://github.com/siemens/ros-sharp/issues?q=is%3Aissue+is%3Aclosed) issues for an answer. I tried my best to find the answer by myself without success. I believe that the discussion we will have in this issue, and the solutions we might find, will help me, and likely other community members who have a similar problem.

**Here is my question:**
What is the use of urdf plugin script? After having uploaded the URDF model, in the inspector the urdf plugins appears but are just texts of the plugins in the urdf file. Is there a way to use the real Gazebo plugin?  

---
## I found a bug! ##
* [ ] I am using the latest ROS# version available here on the master branch.
* [ ] I am adding all required information, code and data files, screenshots and log files so that you can reproduce the problem.

* My OS is: 
* My Unity Version is: 
* My ROS Distribution is: 
* My Build Target Platform is: 

**Here is my bug description:**

Perform the following steps reproduce the bug:

Observed results:

Expected results:

Relevant Code:
```
<!-- Insert code here. Optionally attach code or data files to this issue. -->
```
---
## I have a feature request! ##

**I want to do this:**

**For being able to do it, I wish ROS# had the following feature:**

MartinBischoff, 2020-12-01:
Hi @Tonirayn ,

via the URDF plugin you can import, edit and export the text in Unity. 
Taking a look at the [code](https://github.com/siemens/ros-sharp/blob/master/Unity3D/Assets/RosSharp/Scripts/Urdf/UrdfComponents/UrdfPlugin.cs) we see that it does nothing but reading and writing the text from/to the URDF. Please explain in more detail what functionality you are missing.

Tonirayn, 2020-12-01:
I wish to use the gazebo plugins also in Unity. For example moveit takes advantage of the gazebo_ros_control plugin to allow the manipulator to reach a specific goal leaving the control of the trajectory to ros_control.

MartinBischoff, 2020-12-14:
@Tonirayn

Enabling Unity to use ros_control plugin and behave towards MoveIt as it was Gazebo on the ROS system involves several technical issues for which I don't see a nice solution. One problem we would have to solve is to execute the .so file for the virtual controller on the Unity side though it has been compiled for different OS.

Instead I'd suggest to use the already existing possibilites and re-model the contents of ros_control on the Unity side:
* ROS# provides a topic communication interface with ROS. With this you can interact with MoveIt e.g. via the MoveGroup planning interface.

* gazebo_ros_control plugins are made for loading simulated controller libraries as shared object (.so) files into Gazebo. You can do something similar with Unity by importing a corresponding library file of your controller, e.g. a dll, as a plugin in your Unity project and access this via a C# script, to e.g. execute the controller's step function every FixedUpdate().

Does that answer your question? Do you have any better idea for enabling Unity to understanding gazebo plugins?

MartinBischoff, 2021-01-11:
Hi @Tonirayn , having thought some time about this issue, we still don't see any reasonable way to get ros_control plugin interpreted by Unity.
If anyone finds a possiblity to better incorporate ros_control plugin into Unit, please reopen this issue and let us know.
Best, Martin

closed 2021-01-11 by MartinBischoff