siemens/ros-sharp #368: Re-subscribe all subscriptions at runtime to different robot

Chambana, 2020-12-17:
* [ X] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stack Overflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [X ] I have thoroughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

---
## I have a question! ##

* [X] I searched the [Wiki](https://github.com/siemens/ros-sharp/wiki), [open](https://github.com/siemens/ros-sharp/issues) and [closed](https://github.com/siemens/ros-sharp/issues?q=is%3Aissue+is%3Aclosed) issues for an answer. I tried my best to find the answer by myself without success. I believe that the discussion we will have in this issue, and the solutions we might find, will help me, and likely other community members who have a similar problem.

**Here is my question:**
I have 2 robots streaming identical topics and I need to repeatedly switch between the 2 robots at runtime. The robots each have their own ROS Master.  So, at runtime, I want change ROS#'s ROS master IP and then re-initialize / re-subscribe to my currently subscribed topics (since the ROS master IP was changed). Is there a best-practice for doing this in ROS#?

MartinBischoff, 2020-12-18:
Hi @Chambana ,
I'd suggest to adjust [RosConnector](https://github.com/siemens/ros-sharp/blob/master/Unity3D/Assets/RosSharp/Scripts/RosBridgeClient/RosCommuncation/RosConnector.cs), [UnitySubscriber](https://github.com/siemens/ros-sharp/blob/master/Unity3D/Assets/RosSharp/Scripts/RosBridgeClient/RosCommuncation/UnitySubscriber.cs) and [UnityPublisher](https://github.com/siemens/ros-sharp/blob/master/Unity3D/Assets/RosSharp/Scripts/RosBridgeClient/RosCommuncation/UnityPublisher.cs) (if you have).

To keep code simple, we currently initialize and start them once on `Start()` only.
To realize any type of (dis-/re-)connection application to different ROS systems, just call the corresponding methods when needed.

For switching between two ROS systems in your application, I'd proceed as follows:

First add a List of `UnitySubscribers` and `UnityPublishers` to `RosConnector` via `GetComponents()`.
At runtime  (e.g. (Fixed)Update), when `RosConnector` is triggered to switch to the other ROS system (e.g. via `Switch(string newUrl)`):
* tell all subscribers and publishers to unsubscribe / undavertise and stop their subscription threads
* close connection and corresponding connection thread
* start a new `ConnectAndWait` thread
* tell all subscribers and publishers to subscribe / advertise again

If you find a way to generalize RosConnector and UnitySubscriber / UnityPublisher for a more flexible (dis-)connection at runtime without much increasing code complexity or reducing performance, please let me know.

MartinBischoff, 2021-01-11:
Hi @Chambana , did you make progress on this issue? Did my suggestions help? How did you solve it?

closed 2021-03-09 by MartinBischoff

stevenjj, 2021-05-14:
@MartinBischoff : Hi Martin. Just want to give a small update. I managed to get this feature working with little additional code.
Essentially, the desired behavior is to always reconnect when the ROS master has disconnected. My comments below apply to the release 1.6 tag, but I believe still has utility for the latest versions.

I added a boolean `connectedFlag` within RosConnector that gets changed during OnConnect and OnDisconnect events. Then within the Update step of monobehavior, I check if there  the `connectedFlag` is off if it is, then I perform a `RosSocket.Close()` and rerun the `ConnectAndWait()`thread as well as fire a `onReconnected` event if the connection is successful. The event is only fired by again checking if the `connectedFlag` is now true. I found that using this additional connectedFlag was much faster than using WebSocket's approach for checking connection. Somehow, significant frames are dropped if the Websocket's "is connected" method is used.

I also made the Ros Connector a singleton. This enables an easy way for ROS publishers, subscribers, clients/servers in Unity to listen for the onReconnected event.

We also wrote a small wrapper on the main publisher, subscriber, clients and servers (services and actions) to instead look for the ROS connector singleton (instead of requiring the ROS connector to be on the same GameObject) and automatically perform the advertise/unadvertise subscribe/unsubscribe calls whenever the OnReconnect fires. This design choice assumes that there will only be a single ROS connector in the application, which is common anyway given there is usually only one ROS Master.

The implementation above accomplishes two features:

1. The ROS connector and Unity-specific ROS nodes automatically reconnect if the websocket intermittently drops out.
2. Changing ROS master's is a matter of changing the websocket IP then calling RosSocket.close() within the ROSConnector class . The robustness provided by the implementation automatically reconnects to a new ROS master.

Overall, the actual code implementation was very short and I believe can be easily integrated with any ROS# Unity implementations. Unfortunately, I cannot release the specific implementation I did without significant hassle. But I hope the description above is enough to make it a key feature in future ROS# releases.