siemens/ros-sharp #370: Make this available as (one or more) nuget packages

joaoportela, 2021-01-12:
* [x] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stack Overflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [x] I have thoroughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

## I have a feature request! ##

**I want this available as a [nuget](https://www.nuget.org/) package**

Having this available as nuget package would make it much easier to integrate into a new project and keep it up to date.

MartinBischoff, 2021-01-27:
hi @joaoportela , good suggestion.
I will publish NuGet Packages for [MessageGeneration](https://github.com/siemens/ros-sharp/tree/master/Libraries/MessageGeneration), [RosBridgeClient](https://github.com/siemens/ros-sharp/tree/master/Libraries/RosBridgeClient) and [Urdf](https://github.com/siemens/ros-sharp/tree/master/Libraries/Urdf).

MartinBischoff, 2021-02-06:
done:
https://www.nuget.org/packages/RosSharpMessageGeneration
https://www.nuget.org/packages/RosSharpRosBridgeClient
https://www.nuget.org/packages/RosSharpUrdf

closed 2021-02-06 by MartinBischoff

joaoportela, 2021-02-08:
That's fantastic! :)