siemens/ros-sharp #372: LaserScan Msgs not received in Unity Although the /scan topic has one publisher: gazebo and one subscriber: rosbridge websocket

ZahraaBass, 2021-01-21:
* [ Yes] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stack Overflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [Yes ] I have thoroughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

---
## I have a question! ##

* [ Yes] I searched the [Wiki](https://github.com/siemens/ros-sharp/wiki), [open](https://github.com/siemens/ros-sharp/issues) and [closed](https://github.com/siemens/ros-sharp/issues?q=is%3Aissue+is%3Aclosed) issues for an answer. I tried my best to find the answer by myself without success. I believe that the discussion we will have in this issue, and the solutions we might find, will help me, and likely other community members who have a similar problem.

**Here is my question:**

ZahraaBass, 2021-01-22:
update: it turned out that I rosbridge_suite installed...
so now I can chose a type of msg and ROS# will autogenerate 
[image]

but , still, the Debug.Log line in the ReceiveMessage function of LaserScan Subscriber aren't printing

ZahraaBass, 2021-01-22:
okay i guess I located what the exact problem is.
[image]

The Header in LaserScan.cs is not colored, which means that VS cannot know where the class Header is located...
Only when I open Header.cs and play, the Header word gets colored...
[image]

this means that VS isn't reading msgs which are ros specific, but still i dunno how to solve it.

berkayalpcakal, 2021-01-24:
Hi @ZahraaBass ,

Sorry that I am a bit confused. When you put all the ROS# scripts together with message classes into Unity, do you get a build error, or is the situation as it is in the video link above still valid?

As an alternative way of debugging, you can always import the source code of the [RosBridgeClient](https://github.com/siemens/ros-sharp/tree/master/Libraries/RosBridgeClient) into your Assets instead of importing the [.dll](https://github.com/siemens/ros-sharp/blob/master/Unity3D/Assets/RosSharp/Plugins/RosBridgeClient.dll) file, so that you can fully debug the code.  I suggest you to check the line regarding the [deserialized data](https://github.com/siemens/ros-sharp/blob/master/Libraries/RosBridgeClient/RosSocket.cs#L180) received from rosbridge, which includes error logs if any.

ZahraaBass, 2021-01-24:
Hello @berkayalpcakal, Thank you so much for replying.

I'm sorry for causing this confusion. This [link](url) (https://www.youtube.com/watch?v=L9aU0XW6G1o) is still valid, so no I wasn't getting any error during build. Another update is that I was able to get the ranges from ROS but like one value at a time. So when the msg is just one number Unity seems to have no problem in the ReceiverMessage function, but when the msg is made up of more than one entity, the ReceiveMessage function isnt called - like doesnt even execute a Debug.Log("Hello");

Regarding the alternative way you mentioned, you mean putting Libraries directly into the Assets folder instead of the plugins, like what I did here: https://www.youtube.com/watch?v=pHOIiGMqkMM&feature=youtu.be
I'm not sure if I did it right because I got alot of errors (shown in the video).

And one thing I noticed is that the Time.cs class is defined in the namespace RosSharp.RosBridgeClient.MessageTypes.Std although its not in the MessageTypes folder but in the BaseMessages folder. 

[image]

berkayalpcakal, 2021-01-26:
Hi again @ZahraaBass ,

What I understood from the video is that you placed the library source scripts together with the library dll's, which leads to the errors regarding the duplicate definition of the classes. Instead of  copying and pasting the entire library folder, please try placing only the .cs source scripts into the assets folder.

ZahraaBass, 2021-01-27:
Hello @berkayalpcakal, thanks for the update.
okay I think I did it almost right this time: https://youtu.be/27q1Np9LjOo
I still have 24 errors which are mainly caused by the AssemblyInfo.cs file (there are three files with the same name), for example the errors are like: 
_Assets/Libraries/RosBridgeClient/Properties/AssemblyInfo.cs(31,12): error CS0579: Duplicate 'AssemblyCulture' attribute_

If I delete the AssemblyInfo.cs files in Assets/Libraries/Urdf/Properties and Assets/Libraries/MessageGeneration/Properties, that solved the errors but not sure if this is the right way? Also this would generate  this warning (but it can be cleared):
_Assets/Libraries/RosBridgeClient/Protocols/WebSocketNetProtocol.cs(108,26): warning CS1702: Assuming assembly reference 'System.Threading.Tasks.Extensions, Version=4.2.0.0, Culture=neutral, PublicKeyToken=cc7b13ffcd2ddd51' used by 'System.Threading.Channels' matches identity 'System.Threading.Tasks.Extensions, Version=4.2.0.1, Culture=neutral, PublicKeyToken=cc7b13ffcd2ddd51' of 'System.Threading.Tasks.Extensions', you may need to supply runtime policy_

ZahraaBass, 2021-01-29:
I have an update: https://youtu.be/ILsqY7fEucU

I solved the errors (as I mentioned above) and was able to show the msg on the unity console through adding a Debug.Log line in Rossocket.cs script (as shown in video).

berkayalpcakal, 2021-01-31:
Hello again @ZahraaBass,

I am glad to hear that you made a progress. Could you please write down what went wrong before, and how you figured it out?
If you think that the issue is resolved, please close the issue.

ZahraaBass, 2021-02-01:
first changed the API compatability levelto .Net 4.x.
 (the upper tab, press on Edit – Project Settings – player – other settings, and change the API compatability level to : .Net 4.x)
Then  imported the source code of the RosBridgeClient into my Assets instead of importing the .dll file as in this [video](https://youtu.be/27q1Np9LjOo)
this resulted in some errors (which I mentioned how I solved in the previous comment).
and this [video](https://youtu.be/ILsqY7fEucU) shows where I added the Debug.Log line in Rossocket.cs script.
and Thank you so much for the help! @berkayalpcakal

ZahraaBass, 2021-02-03:
Thanks to issue #377, I found what the real problem was. The LaserScan messages coming from Husky in the PlaypenWorld had some ranges which were 'inf'. So I wrote a node that subscribes to /scan topic to get the LaserScan msgs, changes the 'inf' values to 30 (max range of husky robot) and publishes on a topic (/chatter_). Screenshot below.

[image]

Now the LaserScan Subscriber works and the LaserWriter shows these values in the Inspector window
Ofcourse the solution is mentioned in issue #78 but it seems that I didn't understand the title (new to this JSON serialization thing). Anyways thanks!

closed 2021-02-03 by ZahraaBass