siemens/ros-sharp #374: ROS Unity Videostream - half image

pianita, 2021-01-25:
* [x] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stack Overflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [x] I have thoroughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

---
## I have a question! ##

* [x] I searched the [Wiki](https://github.com/siemens/ros-sharp/wiki), [open](https://github.com/siemens/ros-sharp/issues) and [closed](https://github.com/siemens/ros-sharp/issues?q=is%3Aissue+is%3Aclosed) issues for an answer. I tried my best to find the answer by myself without success. I believe that the discussion we will have in this issue, and the solutions we might find, will help me, and likely other community members who have a similar problem.

**Here is my question:**
Hello,
I am sending a videostream from ROS to Unity. I am using the CameraSubscriber from ROS# example code and successfully subscribe to the topic (receive CompressedImage msgs). But when when applying the image as texture I can only see half the image (as if the data buffer was not loaded fully). The same happens if I send only one image - sometimes the whole image is applied as texture, sometime only half. Maybe someone else had the same problem or any ideas on how I could improve this? 
Thanks!

---

* My Unity Version is: 2020.2.1f1
* My ROS Distribution is: Melodic

MartinBischoff, 2021-01-25:
Hi @pianita please check your [texture / material settings](https://docs.unity3d.com/Manual/Materials.html).

pianita, 2021-01-25:
I did and in the Albedo square I can see that only half the image is displayed as well. Rarely I also get these Errors:

Exception raised!! System.ArgumentOutOfRangeException: Index and count must refer to a location within the buffer.
Parameter name: bytes
at System.Text.UTF8Encoding.GetString (System.Byte[] bytes, System.Int32 index, System.Int32 count) [0x00057] in <9577ac7a62ef43179789031239ba8798>:0
at RosMessageGeneration.Message.DeserializeString (System.Byte[] data, System.Int32 offset, System.Int32 stringLength)

Exception raised!! System.ArgumentOutOfRangeException: Non-negative number required.
Parameter name: count
at System.Text.UTF8Encoding.GetString (System.Byte[] bytes, System.Int32 index, System.Int32 count) [0x0003a] in <9577ac7a62ef43179789031239ba8798>:0
at RosMessageGeneration.Message.DeserializeString (System.Byte[] data, System.Int32 offset, System.Int32 stringLength)

So I guess it has to do with the Deserialization provided by the Unity Robotics Hub. I will change to their repository for help. But thank you for your suggestion!

MartinBischoff, 2021-01-25:
@pianita yes, please ask at Unity Robotics Hub for help if the error is caused in a script that you got there. Closing.

closed 2021-01-25 by MartinBischoff