siemens/ros-sharp #376: Using with Ros2: Ros2webbridge crashes after a period of time

slogan-oxts, 2021-01-26:
* [x] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stack Overflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [x] I have thoroughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

---
## I have a question! ##

* [x] I searched the [Wiki](https://github.com/siemens/ros-sharp/wiki), [open](https://github.com/siemens/ros-sharp/issues) and [closed](https://github.com/siemens/ros-sharp/issues?q=is%3Aissue+is%3Aclosed) issues for an answer. I tried my best to find the answer by myself without success. I believe that the discussion we will have in this issue, and the solutions we might find, will help me, and likely other community members who have a similar problem.

**Here is my question:**

- OS: Windows 10
- Ros: Ros2 Eloquent
- Unity: 2019.4.14f1
- Node.js: 14.15.4

I'm trying to use Ros# with Ros2 using Ros2webbridge.

I get the messages in unity but after a length of time (< 1 minute) after subscribing, Ros2webbridge crashes with the error:
```
#
# Fatal error in , line 0
# Check failed: result.second.
#
#
#
#FailureMessage Object: 000000CFCB4FD850
 1: 00007FF7314F058F napi_wrap+109311
 2: 00007FF731424D8F std::basic_ostream<char,std::char_traits<char> >::operator<<+57151
 3: 00007FF73205E412 V8_Fatal+162
 4: 00007FF731B022AD v8::internal::BackingStore::Reallocate+653
 5: 00007FF731D480E9 v8::ArrayBuffer::GetBackingStore+137
 6: 00007FF7314D2469 napi_get_typedarray_info+393
 7: 00007FF9FA289B41
 8: 00007FF9FA28F28D
 9: 00007FF9FA287EC3
10: 00007FF9FA28EAE3
11: 00007FF7314CCD86 node::Stop+35046
12: 00007FF731D0E72F v8::internal::Builtins::builtin_handle+321471
13: 00007FF731D0DCC4 v8::internal::Builtins::builtin_handle+318804
14: 00007FF731D0DFB7 v8::internal::Builtins::builtin_handle+319559
15: 00007FF731D0DE03 v8::internal::Builtins::builtin_handle+319123
16: 00007FF731DE9F9D v8::internal::SetupIsolateDelegate::SetupHeap+464173
17: 00007FF731D828A2 v8::internal::SetupIsolateDelegate::SetupHeap+40498
18: 00007FF731D828A2 v8::internal::SetupIsolateDelegate::SetupHeap+40498
19: 00000267A1710CFC
```

or sometimes:
```
#
# Fatal error in , line 0
# Check failed: result.second.
#
#
#
#FailureMessage Object: 000000ADA7CFDE50
 1: 00007FF7314F058F napi_wrap+109311
 2: 00007FF731424D8F std::basic_ostream<char,std::char_traits<char> >::operator<<+57151
 3: 00007FF73205E412 V8_Fatal+162
 4: 00007FF731B022AD v8::internal::BackingStore::Reallocate+653
 5: 00007FF731D480E9 v8::ArrayBuffer::GetBackingStore+137
 6: 00007FF7314D2469 napi_get_typedarray_info+393
 7: 00007FF9FDD89B41
 8: 00007FF9FDD8F28D
 9: 00007FF9FDD87EC3
10: 00007FF9FDD8EAE3
11: 00007FF7314CCD86 node::Stop+35046
12: 00007FF731D0E72F v8::internal::Builtins::builtin_handle+321471
13: 00007FF731D0DCC4 v8::internal::Builtins::builtin_handle+318804
14: 00007FF731D0DFB7 v8::internal::Builtins::builtin_handle+319559
15: 00007FF731D0DE03 v8::internal::Builtins::builtin_handle+319123
16: 00007FF731DE9F9D v8::internal::SetupIsolateDelegate::SetupHeap+464173
17: 00007FF731D828A2 v8::internal::SetupIsolateDelegate::SetupHeap+40498
18: 00007FF731D828A2 v8::internal::SetupIsolateDelegate::SetupHeap+40498
19: 00007FF731D828A2 v8::internal::SetupIsolateDelegate::SetupHeap+40498
20: 00007FF731D828A2 v8::internal::SetupIsolateDelegate::SetupHeap+40498
21: 00007FF731DBB475 v8::internal::SetupIsolateDelegate::SetupHeap+272901
22: 00007FF731E5FEFE v8::internal::SetupIsolateDelegate::SetupHeap+947342
23: 00007FF731D828A2 v8::internal::SetupIsolateDelegate::SetupHeap+40498
24: 00007FF731D828A2 v8::internal::SetupIsolateDelegate::SetupHeap+40498
25: 00007FF731D828A2 v8::internal::SetupIsolateDelegate::SetupHeap+40498
26: 00007FF731D7E661 v8::internal::SetupIsolateDelegate::SetupHeap+23537
27: 00007FF731E68343 v8::internal::SetupIsolateDelegate::SetupHeap+981203
28: 00007FF731D828A2 v8::internal::SetupIsolateDelegate::SetupHeap+40498
29: 00007FF731D828A2 v8::internal::SetupIsolateDelegate::SetupHeap+40498
30: 00007FF731D7E661 v8::internal::SetupIsolateDelegate::SetupHeap+23537
31: 00007FF731E68343 v8::internal::SetupIsolateDelegate::SetupHeap+981203
32: 00007FF731D828A2 v8::internal::SetupIsolateDelegate::SetupHeap+40498
33: 00007FF731D828A2 v8::internal::SetupIsolateDelegate::SetupHeap+40498
34: 00007FF731DFC0BE v8::internal::SetupIsolateDelegate::SetupHeap+538190
35: 00007FF731D828A2 v8::internal::SetupIsolateDelegate::SetupHeap+40498
36: 00007FF731D8055E v8::internal::SetupIsolateDelegate::SetupHeap+31470
37: 00007FF731D8014C v8::internal::SetupIsolateDelegate::SetupHeap+30428
38: 00007FF731C50631 v8::internal::Execution::CallWasm+1649
39: 00007FF731C4FE9F v8::internal::Execution::Call+191
40: 00007FF731D3B7E7 v8::Function::Call+615
41: 00007FF7315145FB node::CallbackScope::~CallbackScope+1659
42: 00007FF731514A61 node::MakeCallback+209
43: 00007FF731514BF8 node::MakeCallback+312
44: 00007FF73151496B node::MakeCallback+139
45: 00007FF9ECA47FDD rclnodejs::ShadowNode::Execute+557 [P:\ros2webbridge\ros2-web-bridge\node_modules\rclnodejs\src\shadow_node.cpp]:L123
46: 00007FF9ECA1D59E rclnodejs::Executor::ExecuteReadyHandles+158 [P:\ros2webbridge\ros2-web-bridge\node_modules\rclnodejs\src\executor.cpp]:L258
47: 00007FF73154484B uv_async_send+331
48: 00007FF731543FDC uv_loop_init+1292
49: 00007FF73154417A uv_run+202
50: 00007FF7314502B5 v8::internal::OrderedHashTable<v8::internal::OrderedHashSet,1>::NumberOfBucketsOffset+9365
51: 00007FF7314C3A17 node::Start+311
52: 00007FF73132686C RC4_options+339820
53: 00007FF7322C604C v8::internal::compiler::RepresentationChanger::Uint32OverflowOperatorFor+153532
54: 00007FFA5C1E7C24 BaseThreadInitThunk+20
55: 00007FFA5CF8D4D1 RtlUserThreadStart+33
```

My unity code is looks like this:

```
public class ExampleAction : MonoBehaviour
{
    private RosConnector rosConnector;

    private void Start()
    {
        rosConnector = GetComponent<RosConnector>();

        new Thread(Subscribe).Start();
        new Thread(Subscribe2).Start();

        Debug.Log("Start called");
    }

    private void Subscribe()
    {
        if (!rosConnector.IsConnected.WaitOne(10 * 1000))
            Debug.LogWarning("Failed to subscribe: RosConnector not connected");

        rosConnector.RosSocket.Subscribe<RosSharp.RosBridgeClient.MessageTypes.Std.String>("/topic", SubscriptionHandler, 0); // the rate(in ms in between messages) at which to throttle the topics
    }

    private void Subscribe2()
    {
        if (!rosConnector.IsConnected.WaitOne(10 * 1000))
            Debug.LogWarning("Failed to subscribe: RosConnector not connected");

        rosConnector.RosSocket.Subscribe<RosSharp.RosBridgeClient.MessageTypes.Std.String>("/topic2", SubscriptionHandler, 0); // the rate(in ms in between messages) at which to throttle the topics
    }

    private static void SubscriptionHandler(std_msgs.String message)
    {
        Debug.Log((message).data);
    }
}
```
The topics are being published by a simple python script using rclpy, though I was originally trying to replay a bag file with similar results.

I've tried downgrading Node.js to version 12.9.0.  It lasts much longer using this version (5-10 minutes) but eventually closes with no error.

MartinBischoff, 2021-01-27:
Hi @slogan-oxts ,
did you check issue https://github.com/siemens/ros-sharp/issues/315  and apply the ROS2-related fixes mentioned there?
Maybe the contributors of that issue can help you further?

slogan-oxts, 2021-01-27:
Hi @MartinBischoff

Yes I did try the resolution in that ticket but it seemed to make no difference.

I've managed to get a test using only ros2webbridge working and it seems to fail with the same issue so it looks like the issue is probably in there.  I'll close this one for now and make a ticket on that repository.

closed 2021-01-27 by slogan-oxts

MartinBischoff, 2021-01-27:
Thanks @slogan-oxts , please feel free to reopen if you see that the problem is related to ROS#.
Can you share the link to the other issue here?

slogan-oxts, 2021-01-27:
The solution to this issue was to use a different version of node.js.  The latest erbium release (12.20.1 at the time of writing).

After installing this is seems to be working perfectly both inside and outside of ROS#.

More information in the ros2webbridge ticket:
[github.com/RobotWebTools/ros2-web-bridge/issues/173](github.com/RobotWebTools/ros2-web-bridge/issues/173)