siemens/ros-sharp #377: Turtlebot 2 Laserscan issues

bryceikeda, 2021-02-02:
* [x] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stack Overflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [x] I have thoroughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

---
## I have a question! ##

* [x] I searched the [Wiki](https://github.com/siemens/ros-sharp/wiki), [open](https://github.com/siemens/ros-sharp/issues) and [closed](https://github.com/siemens/ros-sharp/issues?q=is%3Aissue+is%3Aclosed) issues for an answer. I tried my best to find the answer by myself without success. I believe that the discussion we will have in this issue, and the solutions we might find, will help me, and likely other community members who have a similar problem.

**Here is my question:**

I am trying to receive laser scan data from the /scan topic published by my Turtlebot 2. For some reason Unity is not receiving the scans. Interestingly though, a GUI I wrote in ROS displays both my simulated laser scan and the laser scan published by the Turtlebot 2. This is how I know the Turtlebot 2 is publishing data to the correct topic. Unity on the other hand, only displays the laser scan published by my simulation. I put a Debug.Log statement in my Laser Scan Subscriber and it does not get triggered by the Turtlebot 2 /scan data. 

I followed the steps on issue #372 which had an almost similar problem with the laser scan subscriber but it did not fix my problem. I have attached bag files with the broken published /scan data and the working published /scan data. I am using the Laser Scan Subscriber, Laser Scan Visualizer Lines and Laser Scan Writer scripts in my project. Does anyone have any insight as to what the problem is? 

[bagfiles.zip](https://github.com/siemens/ros-sharp/files/5908655/bagfiles.zip)

bryceikeda, 2021-02-02:
I figured out the problem. The /scan messages coming from the Turtlebot had a bunch of 'nan' ranges. I created a new topic that published a laserscan message that switched all the 'nan' ranges to 0.0. Now the Laser Scan Subscriber receives the message properly and displays them in Unity.

closed 2021-02-02 by bryceikeda

MartinBischoff, 2021-02-03:
This issue has been discussed in this board multiple times already, e.g. https://github.com/siemens/ros-sharp/issues/78 and https://github.com/siemens/ros-sharp/issues/334

ZahraaBass, 2021-02-03:
Hello @bryceikeda, would you mind sharing the script you wrote for switching the "nan" ranges?

bryceikeda, 2021-02-04:
@ZahraaBass I just for loop through all of the ranges in /scan and use isnan() to figure out if it is nan. If it is, I replace it with 0.0. Once that is done I publish the fixed scan variable over a new topic such as /fixed_scan