siemens/ros-sharp #378: ROS BAG file parsing?

kirk-marple, 2021-02-28:
* [X] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stack Overflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [X] I have thoroughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

---
## I have a question! ##

* [X] I searched the [Wiki](https://github.com/siemens/ros-sharp/wiki), [open](https://github.com/siemens/ros-sharp/issues) and [closed](https://github.com/siemens/ros-sharp/issues?q=is%3Aissue+is%3Aclosed) issues for an answer. I tried my best to find the answer by myself without success. I believe that the discussion we will have in this issue, and the solutions we might find, will help me, and likely other community members who have a similar problem.

**Here is my question:**
I've been reading through the code and issues, and checked out the wiki, but still have a question.

Does ros-sharp support ROS BAG file parsing?  I simply want to be able to subscribe to topics in a ROS BAG, and export the data samples for further processing.  (i.e. pulling out video frames within a .NET application)

I've done this before using ROS directly, but I can't tell if this is inside or outside the scope of ros-sharp?

Thanks!

berkayalpcakal, 2021-03-03:
Hello @kirk-marple,

rosbag file parsing is not supported by ROS#, i.e. there is no way to decode rosbag data within the .NET environment via ROS#. 

rosbag files are encoded within the ROS environment via the rospackage [rosbag](http://wiki.ros.org/rosbag). It is not a standalone tool to be included in a non-ROS platform, because it depends on core ros libraries (the  rosbag package publishes the decoded messages directly to the ROS environment as it reads the file).

It should be possible to implement a C# script to read/write rosbag files by following the [bag format](http://wiki.ros.org/Bags/Format) specifications. If you want to come up with any implementation, you might find [this documentation](https://docs.ros.org/en/api/rosbag/html/c++/) and [this repo](https://github.com/ros/ros_comm/tree/noetic-devel/tools/rosbag_storage) useful.
 
I hope this helps. 
If you would have an idea for getting such a feature in ROS#, please do not hesitate to post your comments here.

closed 2021-03-03 by berkayalpcakal

kirk-marple, 2021-03-03:
Really appreciate the info, thanks.    I think it's something we'll look into later this year, and I'll definitely keep an eye on how to best fit this into ROS#, since I'm sure others would find it useful.