siemens/ros-sharp #380: file_server.launch missing

ruanViljoen, 2021-03-09:
* [x] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stack Overflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [x] I have thoroughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

---
## I have a question! ##

* [ x] I searched the [Wiki](https://github.com/siemens/ros-sharp/wiki), [open](https://github.com/siemens/ros-sharp/issues) and [closed](https://github.com/siemens/ros-sharp/issues?q=is%3Aissue+is%3Aclosed) issues for an answer. I tried my best to find the answer by myself without success. I believe that the discussion we will have in this issue, and the solutions we might find, will help me, and likely other community members who have a similar problem.

**Here is my question:**
I am trying to import a URDF model into Unity from ROS, but the required file_server.launch file seems to be missing? The link provided gives a 404. https://github.com/siemens/ros-sharp/tree/master/ROS/file_server/launch/file_server.launch

MartinBischoff, 2021-03-09:
Hi @ruanViljoen,
there is no `file_server.launch`.
You can start the `file_server` node together with rosbridge WebSocket via [`ros_sharp_communication.launch`](https://github.com/siemens/ros-sharp/blob/master/ROS/file_server/launch/ros_sharp_communication.launch), i.e.
```roslaunch file_server ros_sharp_communication.launch```
Please let me know where you found the 404 link above such that we can correct it.

ruanViljoen, 2021-03-09:
Thank you, it works now. The link is in the ROS/README.md readme file.

Is there existing functionality in the ros-sharp library to animate the URDF based on incoming /tf messages, or will I have to implement that myself?

MartinBischoff, 2021-03-09:
Fixed in commit https://github.com/siemens/ros-sharp/commit/3054f8844df7c463830e2d57b6750ab20bde9bd8

Concerning your other question please read the Wiki pages [GazeboSimulationExample](https://github.com/siemens/ros-sharp/wiki/User_App_ROS_GazeboSimulationExample) and [AnimateRobotModelInUnity3D](https://github.com/siemens/ros-sharp/wiki/User_App_NoROS_AnimateRobotModelInUnity3D).
Rather than `\tf` I'd suggest you update Unity transforms via `\joint_states` for which corresponding Unity scripts aready exist.

closed 2021-03-09 by MartinBischoff