siemens/ros-sharp #381: Regarding instantiation of gameobjects inside service response handler

karthikm-0, 2021-03-12:
Hi,

I am trying to take a set of waypoints generated by a ROS service and visualize them in Unity using a prefab. However, I'm finding that invoking Instantiate inside a for loop of waypoints from the service response causes it to stop after the first entry of the loop (the first waypoint). I'm very confused as to why this is happening:

Here's the function that is called from the response handler:

`    public List<GameObject> VisualizePoses(RosSharp.RosBridgeClient.MessageTypes.Geometry.Pose[] poses, string poseType)
        {
            // Create gameobjects
            var toVisualize = new List<GameObject>();
        
            Debug.Log("Visualizing trajectory waypoints.");
            foreach (var w in poses)
            {
                var pos = new Vector3((float) w.position.x, (float) w.position.y, (float) w.position.z);
                var ori = new Quaternion((float) w.orientation.x, (float) w.orientation.y, 
                    (float) w.orientation.z, (float) w.orientation.w);
                
                if (poseType == "saved")
                {
                    pos = pos.Ros2Unity();
                    ori = ori.Ros2Unity();
                    Debug.Log(pos);
                    Debug.Log(ori);
                    var wPoint = Instantiate(trajPoint, pos, ori);
                    toVisualize.Add(wPoint);
                }
            }
            
            return toVisualize;`

Uncommenting the Instantiate invocation allows me to see all the other waypoints (the two debug statements) but when calling it, the function stops at this point and does not spawn the prefab or continue the loop.

MartinBischoff, 2021-03-15:
Hi @karthikm-0 ,

is there an error message in your console? Can you please post it here?

Callbacks are typically executed upon message receival which is outside of Unity's main thread.
* In your Callback just save the pos and ori in a `Vector[]` and `Quaternion[]` and set a `boolean doUpdate` to `true`.
* In your `Update()`: `if (doUpdate)`, instantiate all your trajectory waypoints using your `pos` an `ori` arrays and set `doUpdate=false`.

karthikm-0, 2021-03-15:
Hi @MartinBischoff,

Thanks for the response. I have no error messages in the console. I ended up using a workaround similar to what you said where I set a bool to true when the message is received and then instantiate the objects. However, I want to avoid doing this if possible, since I'm not actually using an Update function so I need coroutines each time I call a service and use a while loop there until the response is received. I was wondering what is causing this behavior because it seems that the callback expires after some set time which would explain why the objects are not instantiated inside this scope. I was recently trying out the Unity Robotics Hub's implementation and was able to instantiate within the callback so that's why I'm particularly curious.

MartinBischoff, 2021-03-15:
Without having seen the environment in which you call the method posted above, I guess that the reason is that (in ROS#) the callback is executed in a separate thread, and not in Unity's main thread.
This has the advantage that things can be prepared and pre-calculated right upon message receival to minimize effort next time Unity's main thread is called.

`Instantiate` works in Unity's main thread only and causes the strange behaviour (see e.g. [here](https://answers.unity.com/questions/795494/create-gameobjects-in-threads.html))

Of course you can use your (Unity) Coroutines instead of `Update()` to process the data you received in the callback. They run in Unity's main thread again [see e.g. here](https://support.unity.com/hc/en-us/articles/208707516-Why-should-I-use-Threads-instead-of-Coroutines-).

Out of curiousity: has your `trajPoint` been instantiated at the first pose you received or at the last one?

karthikm-0, 2021-03-15:
@MartinBischoff I had a suspicion this was thread-related, so this makes perfect sense, thanks! trajPoint is the prefab instance I assign in the Editor or can load with Resources.Load. The goal is to instantiate one for each pose after all poses have been received.

closed 2021-03-16 by MartinBischoff