siemens/ros-sharp #383: Not connecting to Rosbridge tcp server with BSON

MrP0w3r, 2021-03-18:
I have the publisher and subscriber example working with JSON and websocket, connecting to rosbridge: 

rosbridge_server rosbridge_websocket.launch.

However I have to run rosbridge as TCP BSON:

rosbridge_server rosbridge_tcp.launch bson_only_mode:=True

It will connect with "Newtonsoft_BSON" and "Web Socket Sharp" but not publish or subscribe.

(rosbridge 0.9.0)
(ROS Melodic)
(Unity 2019.4.20f1)

berkayalpcakal, 2021-03-24:
Hello @MrP0w3r,

Could you please elaborate more on the error you get? What do you mean by not connecting properly?

I launched the rosridge server with bson_only_mode on the ROS side, and set the serializer "Newtonsoft_BSON" on the Unity side. It seems that Unity connects to rosbridge, however, when I subscribe to a topic (there is no message flow on the topic, I am just sending [subscription object](https://github.com/RobotWebTools/rosbridge_suite/blob/develop/ROSBRIDGE_PROTOCOL.md#344-subscribe) to rosbridge) I observe the following error;

[image]

I do not know much about this "UnicodeDecodeError: 'ascii' codec can't decode byte ... " error, but my guess is that the BSON objects are encoded/decoded with 'utf-8' encoding type on the rosbridge side whereas it is 'ascii' on the ROS# side. But maybe I am wrong, I have not worked with BSON before.

BSON support is added to ROS# with #198 by @akirayou, maybe he can help at this point.

MartinBischoff, 2024-01-22:
we are planning to remove BSON in future releases.

closed 2024-01-22 by MartinBischoff