siemens/ros-sharp #391: Not all resource files have been received before timeout Unity ROS#

medinags, 2021-04-29:
**I can't import my robot**
Hello, 

My OS is: VM Ubuntu 18.04
My Unity Version is: 2019.4.23f1
My ROS Distribution is: ROS-melodic

I am trying to import my robot from ROS to Unity but, I find the following error in Unity. 

At first everything looks fine, but then this error appears. 

[image]

On the ROS side I encounter the following problem

[image]

```
[id: /file_server/get_file:1] call_service ServiceException: transport error completing service call: 
receive_once[/file_server/get_file]: DeserializationError cannot deserialize: unknown error handler name 'rosmsg'
```
---
I don't know what it is, I have tried to solve it, but I have not been able to. 
I have checked my .urdf file and everything seems to be correct.
Thank you for your help

AmyPhung, 2021-08-09:
I also have this issue, and have tried increasing the timeout to no avail. Has anyone found a fix for this? 

My OS is: Ubuntu 18.04
My Unity Version is: 2020.3.15f2
My ROS Distrubution is: ROS melodic

MartinBischoff, 2021-08-10:
Hi @AmyPhung, which resource files are used in your urdf? Can you share it here?

AmyPhung, 2021-08-10:
Sure! I'm trying to load [kraft.urdf](https://bitbucket.org/droplabumich/kraft_arm_pkgs/src/master/kraft_arm_description/robot/kraft.urdf), and the mesh files are stored [here](https://bitbucket.org/droplabumich/kraft_arm_pkgs/src/master/kraft_arm_description/meshes/KRAFTARM/). After starting up the fileserver and rosbridge server, I'm using [this launch file](https://bitbucket.org/droplabumich/kraft_arm_pkgs/src/master/kraft_arm_moveit_config/launch/demo.launch) to load the robot description into ROS. Unlike @medinags though, when I click the "Read Robot Description" button in Unity, my fileserver doesn't report any `get_file` requests - I only have the `ROSbridge File Server initialized` message

MartinBischoff, 2021-08-10:
Hi @AmyPhung , what is the error message your receive? Which import steps are done / open?
It first glance it looks to me like to did not set `robot/name` in your `launch` file. Please take a look at [publish_description_turtlebot2.launch](https://github.com/siemens/ros-sharp/blob/master/ROS/file_server/launch/publish_description_turtlebot2.launch) for reference.

AmyPhung, 2021-08-10:
I get the error `Not all resource files have been received before timeout`. The `Connected`, `Robot Name Received`, `Robot Description Received`, and `Disconnected` are marked as done, and the `Resource Files Received` and `Import Complete` are marked as open.

[image]

MartinBischoff, 2021-08-11:
@AmyPhung Please try and resolve the ROS error `unknown-error-handler-name-rosmsg`:
https://answers.ros.org/question/360537/unknown-error-handler-name-rosmsg/

Your ROS system can not interpret the file_server get_file server requests due to this error.

AmyPhung, 2021-08-11:
Thanks for pointing that out, I didn't notice that error. I got it to work, I'm not sure what exactly fixed it but I did the following:
+ `sudo apt update` 
+ `sudo apt upgrade`
+ `sudo add <myuser> dialout`
+ Re-sourcing the workspace before running the rosbridge

I think it was the last item that ultimately got it to work, but I'll include everything I tried here just in case one of the other things fixed it.

Thanks for the help!

closed 2021-08-11 by MartinBischoff