siemens/ros-sharp #392: LaserScanSubscriber doesn´t work

mrmanu, 2021-05-03:
* My OS is: 
* My Unity Version is: 
* My ROS Distribution is: 
* My Build Target Platform is: 

**Here is my bug description:**

Perform the following steps reproduce the bug:

Observed results:

Relevant Code:
```
<!-- namespace RosSharp.RosBridgeClient
{
    public class LaserScanSubscriber : UnitySubscriber<MessageTypes.Sensor.LaserScan>
    {
        public LaserScanWriter laserScanWriter;

        protected override void Start()
        {
            base.Start();
        }

        protected override void ReceiveMessage(MessageTypes.Sensor.LaserScan laserScan)
        {
            print("AAAA");
            laserScanWriter.Write(laserScan);
        }
    }
} -->
```
---

MartinBischoff, 2021-05-04:
Hi @mrmanu ,
your issue seems to duplicate https://github.com/siemens/ros-sharp/issues/346. I found two suggestions there:
* replace [System.Threading.Tasks.Extensions.dll](https://github.com/siemens/ros-sharp/issues/346#issuecomment-712018330)
* replace [ROS bridge python client](https://github.com/siemens/ros-sharp/issues/346#issuecomment-703171779)

mrmanu, 2021-05-04:
Hello @MartinBischoff ,  thanks for the reply.
I think my problem is different from those, however I have tried these solutions and they do not work for me.
I have checked the RosBridge protocol and it should be correct since all other topics are subscribed correctly. I have also replaced System.Threading.Tasks.Extensions.dll with the latest updated one and it gives no results.
I think the problem must be in LaserScan in particular since all the other topics work correctly, and I have even created new types (to subscribe to the pcl/PolygonMesh topic) and they work fine.

berkayalpcakal, 2021-05-05:
Hello @mrmanu,

Have you checked the issues #78, #334, #377? This could be a known issue regarding the serialization of NaN/inf values within the laserscan message. 

If you think that the issue is not related to aforementioned issues, maybe you can try out the [RosBridgeConsoleExample](https://github.com/siemens/ros-sharp/blob/master/Libraries/RosBridgeClientTest/RosSocketConsoleExample.cs) by adjusting it in a way that you would only have a laserscan subscriber. You can just open the Visual Studio solution under the `Libraries` folder and run the console example. This would help you debug every step of the handling of incoming messages, e.g. you can check what happens then [this line](https://github.com/siemens/ros-sharp/blob/master/Libraries/RosBridgeClient/RosSocket.cs#L180) is being executed.  

I think it is not possible to reproduce your issue with the current info you provided. I really suggest you to debug the library thoroughly to find out an error.

closed 2021-08-10 by MartinBischoff