siemens/ros-sharp #393: Is the websocket class inclued in offcial library ?(means that actually you dont implememnt the websocket?)

gaoxiaohuicoco, 2021-05-14:
* [ ] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stack Overflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [ ] I have thoroughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

---
## I have a question! ##

* [ ] I searched the [Wiki](https://github.com/siemens/ros-sharp/wiki), [open](https://github.com/siemens/ros-sharp/issues) and [closed](https://github.com/siemens/ros-sharp/issues?q=is%3Aissue+is%3Aclosed) issues for an answer. I tried my best to find the answer by myself without success. I believe that the discussion we will have in this issue, and the solutions we might find, will help me, and likely other community members who have a similar problem.

**Here is my question:**

 public WebSocketSharpProtocol(string url)
        {
            WebSocket = new WebSocket(url);
            WebSocket.OnMessage += Receive;

            WebSocket.OnClose += Closed;
            WebSocket.OnOpen += Connected;
        }
I want to know about  websocket class details,but i cant find it
---
## I found a bug! ##
* [ ] I am using the latest ROS# version available here on the master branch.
* [ ] I am adding all required information, code and data files, screenshots and log files so that you can reproduce the problem.

* My OS is: 
* My Unity Version is: 
* My ROS Distribution is: 
* My Build Target Platform is: 

**Here is my bug description:**

Perform the following steps reproduce the bug:

Observed results:

Expected results:

Relevant Code:
```
<!-- Insert code here. Optionally attach code or data files to this issue. -->
```
---
## I have a feature request! ##

**I want to do this:**

**For being able to do it, I wish ROS# had the following feature:**

gaoxiaohuicoco, 2021-05-14:
i cant find websocet class in your project,thank you,bro

MartinBischoff, 2021-05-14:
Hi @gaoxiaohuicoco ,
the plugin we are using in this script is contained in the Unity project [here](https://github.com/siemens/ros-sharp/blob/master/Unity3D/Assets/RosSharp/Plugins/External/websocket-sharp.dll).
You find the code of that plugin [here](https://github.com/sta/websocket-sharp).

closed 2021-05-14 by MartinBischoff

gaoxiaohuicoco, 2021-05-14:
thank you，bro