siemens/ros-sharp #394: How can I navigate in unity using ros#?

837910164, 2021-05-14:
* [ ] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stack Overflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [ ] I have thoroughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

---
## I have a question! ##

* [ ] I searched the [Wiki](https://github.com/siemens/ros-sharp/wiki), [open](https://github.com/siemens/ros-sharp/issues) and [closed](https://github.com/siemens/ros-sharp/issues?q=is%3Aissue+is%3Aclosed) issues for an answer. I tried my best to find the answer by myself without success. I believe that the discussion we will have in this issue, and the solutions we might find, will help me, and likely other community members who have a similar problem.

**Here is my question:**

---
## I found a bug! ##
* [ ] I am using the latest ROS# version available here on the master branch.
* [ ] I am adding all required information, code and data files, screenshots and log files so that you can reproduce the problem.

* My OS is: 
* My Unity Version is: 
* My ROS Distribution is: 
* My Build Target Platform is: 

**Here is my bug description:**

Perform the following steps reproduce the bug:

Observed results:

Expected results:

Relevant Code:
```
<!-- Insert code here. Optionally attach code or data files to this issue. -->
```
---
## I have a feature request! ##

**I want to do this:**

**For being able to do it, I wish ROS# had the following feature:**

837910164, 2021-05-14:
I can get the map by move the robot,but what should i do to navigate in unity?

MartinBischoff, 2021-05-18:
Hi @837910164 !
congratulations! This is the point where your implementation work begins.

You can now use ROS# to communicate from Unity to ROS how you want to navigate your robot.
You can e.g. use MoveIt Motion Planning API to tell your robot to navigate via topics and actions (i.e. [MoveGroup Action](http://docs.ros.org/en/melodic/api/moveit_msgs/html/action/MoveGroup.html).

Concerning the ROS-Unity communication interface ROS# everything works as for receiving the map: just communicate other topics now. For further help concerning navigation please refer to the respective projects.

closed 2021-05-18 by MartinBischoff