siemens/ros-sharp #403: Subscriber stops after a few messages

florianspy, 2021-07-20:
I am using the following source code for a subscriber:
﻿using System;
using UnityEngine;

namespace RosSharp.RosBridgeClient
{
    public class StringSubscriber : UnitySubscriber<MessageTypes.Std.String>
    {
        public String messageData;

        protected override void Start()
        {
            base.Start();
        }

        protected override void ReceiveMessage(MessageTypes.Std.String message)
        {
            messageData = message.data;
            //Debug.Log(messageData);
        }
	
    }
}
However after about 80 message no new data is received. Even though data is still published.
I also tried the following code with the same issue.
using System;
using RosSharp.RosBridgeClient;
using UnityEngine;
using System.Collections;
using System.Collections.Generic;
using RosSharp.RosBridgeClient.Protocols;
// commands on ROS system:
// roslaunch rosbridge_server rosbridge_websocket.launch
// rostopic echo /talker
// rostopic pub /listener std_msgs/String "World!"

public class HelloWorld: MonoBehaviour

{
int count;

public Protocol protocol;
void Start()
{ RosSocket.SerializerEnum serializer = RosSocket.SerializerEnum.Microsoft;
	
RosSocket rosSocket = new RosSocket(ProtocolInitializer.GetProtocol(protocol, "ws://172.16.24.128:9090"),serializer);
    // Subscription:
    rosSocket.Subscribe<RosSharp.RosBridgeClient.MessageTypes.Geometry.PoseStamped >("/listener2",subscriptionHandler);
}
private static void subscriptionHandler(RosSharp.RosBridgeClient.MessageTypes.Geometry.PoseStamped message)
{

	Debug.Log("Hello message"+message);
}
   void Update()
    {
	//Debug.Log("subscriber");
count=count+1;
/*	if(count>100){
		count=0;
		rosSocket.Close();
		RosSocket.SerializerEnum serializer = RosSocket.SerializerEnum.Microsoft;
		rosSocket = new RosSocket(ProtocolInitializer.GetProtocol(protocol, "ws://172.16.24.128:9090"),serializer);
    		rosSocket.Subscribe<RosSharp.RosBridgeClient.MessageTypes.Geometry.PoseStamped >("/listener2",subscriptionHandler);
	}*/
    }
}

florianspy, 2021-07-21:
After creating the problem again from scratch using the rossharp as base of the project https://github.com/siemens/ros-sharp/tree/master/Unity3D. The problem is solved.

MartinBischoff, 2021-07-21:
Thanks for the info @florianspy . Glad to hear this issue is resolved.
In case you have any further info about the actual reason, please do share it here as it might help others have the same issue.

closed 2021-07-21 by MartinBischoff