siemens/ros-sharp #404: "URDF Model (New)" Have No Rigidbodies, New Links Have No Joints

kneave, 2021-07-29:
* [X] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stack Overflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [X] I have thoroughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

---
## I have a question! ##

* [X] I searched the [Wiki](https://github.com/siemens/ros-sharp/wiki), [open](https://github.com/siemens/ros-sharp/issues) and [closed](https://github.com/siemens/ros-sharp/issues?q=is%3Aissue+is%3Aclosed) issues for an answer. I tried my best to find the answer by myself without success. I believe that the discussion we will have in this issue, and the solutions we might find, will help me, and likely other community members who have a similar problem.

**Here is my question:**

Following the ("create a new URDF")[https://youtu.be/07q2mGxD2j8] tutorial I found that none of the new links had joints created. In the video no extra steps were needed for this but I found I needed to manually add a rigidbody to the base_link in order for joints to be created in the manner shown in the video. I couldn't find reference to adding the rigidbody in the wiki or previous issues. Is this change by design or a regression of some kind?

---
## I found a bug! ##
* [X] I am using the latest ROS# version available here on the master branch.
* [X] I am adding all required information, code and data files, screenshots and log files so that you can reproduce the problem.

* My OS is: Windows 11, dev preview
* My Unity Version is: 2020.3.11f1
* My ROS Distribution is: N/A
* My Build Target Platform is: N/A

**Here is my bug description:**

Creating a new URDF doesn't include a rigidbody on the base_link, this prevents joints being created.

Perform the following steps reproduce the bug:

1. Create a new 3d object by selecting "urdf (new)" in the 3d objects menu
2. Click on the base_link and add a new link 
3. Look at the newly created link, there will be no joint present

Observed results:

Base_link on a new model with no rigidbody
[image]

Newly created child link, no joint component
[image]

Expected results:

Base_link with rigidbody
[image]

New children have joints
[image]

MartinBischoff, 2021-08-02:
Hi @kneave ! is there an inertia tag specified in the URDF, something like
```
<inertia ixx="1"  ixy="0"  ixz="0" iyy="1" iyz="0" izz="1" />
```
This is required to generate  a rigidbody in Unity, see also https://github.com/siemens/ros-sharp/issues/19.

kneave, 2021-08-02:
Hi Martin, this is from creating a new model completely in Unity so no URDF imported in this case. If it's expected behaviour that the user adds a rigid body could I suggest the video demo be updated to show this please or it be included in a the documentation?

MartinBischoff, 2021-08-03:
Hi @kneave , I digged a bit deeper into this issue. We changed the functionality in commit https://github.com/siemens/ros-sharp/commit/f15f3f9eeec0d46b7dbb3dfbc4364eeef8e7fbe8
to not automatically generate rigidbodies in this case.

The problem is that URDF does not require `<inertial>` tags (i.e. rigidbody data) whereas Unity requires rigidbodies to create joints between transforms (i.e. links). We had discussions whether to create a default rigidbody to a new/empty link or not since the beginning and didn't find a clear answer yet.

In pull request #405 I changed it back, such that creating a new link in Unity also creates a rigidbody / default inertial data. 
It seems like this is the more convenient / user friendly solution.
Would you please test if it works for you?

kneave, 2021-08-04:
I just pulled that branch and it looks like it works as I'd expect from the demo video, thanks for the quick update! Good call with the Newtonsoft.Json fix too. :)

MartinBischoff, 2021-08-05:
Merged in https://github.com/siemens/ros-sharp/commit/6e91309b2447c0919d3dabaa0650bb71e3e490dd . Thank you @kneave .

closed 2021-08-05 by MartinBischoff