siemens/ros-sharp #406: after applying the Joint state, all the joints are moving in a wrong way.

fanke1213812138, 2021-09-03:
I am using the ROS# to simulate PSM of the da Vinci robot. I applied joint_state_subscriber, joint_state_writer. I am sure I can receive the Jointstate from rostopic, but all the joints move in a wrong way.
[image]

MartinBischoff, 2021-09-06:
Hi @fanke1213812138 , with 'wrong' you mean opposite, I presume?
Please check our URDF whether they satisfy  the ROS Unity Coordinate System Conventions [here](https://github.com/siemens/ros-sharp/wiki/Dev_ROSUnityCoordinateSystemConversion).

fanke1213812138, 2021-09-06:
Thank you for your help. I think the Coordinate System is correct, since the axis in Unity is already converted as shown in Hinge Joint etc.. I mean when receiving JointState message, some Joints are just separated from their parent joints, as shown below.
[image]

HelloWorldLTY, 2022-04-09:
> Thank you for your help. I think the Coordinate System is correct, since the axis in Unity is already converted as shown in Hinge Joint etc.. I mean when receiving JointState message, some Joints are just separated from their parent joints, as shown below. [image]

Hi, did you try the sample unity program provided by the tutorial? I found that now it cannot work. Thanks a lot!

MartinBischoff, 2024-01-22:
outdated / closing due to inactivity

closed 2024-01-22 by MartinBischoff