siemens/ros-sharp #416: Not able to call service from Unity

gianmarco96, 2021-12-21:
* [x] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stack Overflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [x] I have thoroughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

---
## I have a question! ##

* [x] I searched the [Wiki](https://github.com/siemens/ros-sharp/wiki), [open](https://github.com/siemens/ros-sharp/issues) and [closed](https://github.com/siemens/ros-sharp/issues?q=is%3Aissue+is%3Aclosed) issues for an answer. I tried my best to find the answer by myself without success. I believe that the discussion we will have in this issue, and the solutions we might find, will help me, and likely other community members who have a similar problem.

**Here is my question:**

Hello, thank you for the fantastic work you did with library!

I'm really struggling to call services from Unity. I have checked the wiki and the examples, but they do not seem to work for me. I'm sure I'm being thick and don't think it's necessarily a ROS# issue. I don't really understand the example of call service shown below:
```
rosSocket.CallService<rosapi.GetParamRequest, rosapi.GetParamResponse>("/rosapi/get_param", ServiceCallHandler, new rosapi.GetParamRequest("/rosdistro", "default"));
```
Or better I tried doing the same exact thing to no result. The server is running on the ROS side and when I call it from unity nothing seems to happen. I have created the ServiceCallHandler and generated the service request and response as per instruction. I could provide the example of my application, but I thought that would be very specific to my use case, while instead a more generic example could be useful to everyone. 

Let's say I wanted to call the /spawn service from  turtlesim node, which everyone has access to, Unity using this library. What would I need to do? Possibly step by step? 

I have used this library for over a year now, but never managed to use a service properly (I always had to find a work around using topics), I reckon I'm not the only one so I'm sure a more generic guide would be useful to others as well.

Thank you so much for your help,

gianmarco96, 2021-12-21:
So ok I sorted it by using RosSocket rosSocket = new RosSocket(new RosBridgeClient.Protocols.WebSocketSharpProtocol(uri)); rather than WebSocketNetProtocol, is there a reason why the latter would not work?

MartinBischoff, 2024-01-22:
outdated / closing due to inactivity