siemens/ros-sharp #418: View mesh_msgs::TriangleMeshStamped messages on hololens2

abhid28, 2022-02-11:
Issue Template © Siemens AG, 2017-2018
Author: Dr. Martin Bischoff (martin.bischoff@siemens.com)
-->

* [ ] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stack Overflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [ ] I have thoroughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

---
## I have a question! ##

* [ ] I searched the [Wiki](https://github.com/siemens/ros-sharp/wiki), [open](https://github.com/siemens/ros-sharp/issues) and [closed](https://github.com/siemens/ros-sharp/issues?q=is%3Aissue+is%3Aclosed) issues for an answer. I tried my best to find the answer by myself without success. I believe that the discussion we will have in this issue, and the solutions we might find, will help me, and likely other community members who have a similar problem.

**Here is my question:**
I need help with visualizing mesh_msgs::TriangleMeshStamped messages in hololens 2. I can see them in RVIZ and I am able to publish them. I need to render/ view them on hololens 2 and I am not able to figure out how

---
## I found a bug! ##
* [ ] I am using the latest ROS# version available here on the master branch.
* [ ] I am adding all required information, code and data files, screenshots and log files so that you can reproduce the problem.

* My OS is: <Windows 10>
* My Unity Version is: <2019.4.28f1>
* My ROS Distribution is: <melodic>

EricVoll, 2022-03-14:
I did that once. Here's the code file
https://github.com/EricVoll/ethz_asl_semester_project/blob/main/MrDrone.Unity254/Assets/MeshController/MeshGenerator.cs

MartinBischoff, 2024-01-22:
outdated / closing due to inactivity

closed 2024-01-22 by MartinBischoff