siemens/ros-sharp #422: Image Publisher doesn't work with ROS Image View

jntlblv1, 2022-03-18:
* [ ] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stack Overflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [ ] I have thoroughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

---
## I have a question! ##

* [ ] I searched the [Wiki](https://github.com/siemens/ros-sharp/wiki), [open](https://github.com/siemens/ros-sharp/issues) and [closed](https://github.com/siemens/ros-sharp/issues?q=is%3Aissue+is%3Aclosed) issues for an answer. I tried my best to find the answer by myself without success. I believe that the discussion we will have in this issue, and the solutions we might find, will help me, and likely other community members who have a similar problem.

**Here is my question:**
I'm using Image Publisher and it works partially. 
This is my case: first of all I'm using Image Subscriber in order to get a CompressedImage from ROS and it works.
I publish the image on a Plane object and I add a simple cube. Then I want to see this scene in ROS (e.g. with rqt Image View).
In order to do that I'm using Image Publisher. In ROS I can see the topic but when I try to see my scene with ImageView I can only see a grey screen.

I also tried to subscribe to the topic which i created with another Image Subcriber and I published the image on another plane and it works, so I think that the problem is in ROS.

Can you give me some advice? Thank you very much!

MartinBischoff, 2022-03-18:
Hi @jntlblv1 
Have you checked the hints suggested in closed issues related to image publishers & subscribers, eg. https://github.com/siemens/ros-sharp/issues/352 and referenced?
The problem could be related to codecs used by Unity not available on your ROS system or image data de/-serialization in general. Save the message byte[] in a file, check in in Hex Editors, and image viewers, also on the machine running Unity.

MartinBischoff, 2024-01-22:
outdated / closing due to inactivity

closed 2024-01-22 by MartinBischoff