siemens/ros-sharp #429: Update releases to use Semver

JohnClema, 2022-05-20:
* [x] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stack Overflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [x] I have thoroughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

**I want to do this:**
ros-sharp was just published to openupm [here](https://openupm.com/packages/com.siemens.ros-sharp/), however it requires specific versioning in order to be recognised and pulled down using the [openupm-cli](https://openupm.com/docs/getting-started.html)

**For being able to do it, I wish ROS# had the following feature:**
It was versioned according to [semver](https://semver.org/), openupm requires this as described in package criteria [here](https://openupm.com/docs/adding-upm-package.html#upm-package-criteria). This could be achieved by re releasing v1.7 as 1.7.0 or similar.

JohnClema, 2022-05-20:
@MartinBischoff is this something that might be possible in future?

MartinBischoff, 2022-05-23:
Hi @JohnClema , I just changed all existing releases to refering to semver tags. Please confirm if openupm works for you now.
The old non-semver tags are still there. I think about deleting them...

JohnClema, 2022-05-24:
Hi @MartinBischoff, it isn't working at the moment due to the old tags. If you see here, the build pipelines are failing for the new ones you published: https://openupm.com/packages/com.siemens.ros-sharp/?subPage=pipelines

I've put in a request over here for openupm to unpublish the old non-semver tags. Let me know what you think!
https://github.com/openupm/openupm/issues/3052

MartinBischoff, 2022-05-24:
done - I just deleted all old non-semver tags.

JohnClema, 2022-05-24:
Thanks @MartinBischoff I'll close this one off and continue finalising this over at https://github.com/openupm/openupm/issues/3052

closed 2022-05-24 by JohnClema