siemens/ros-sharp #435: How do you get your desired urdf to be published as robot_description to the ROS parameter server?

kalepo, 2022-07-13:
I'm trying to edit the launch file and replace the turtlebot urdf file and package with a desired urdf file and package. In this case it is a ABB 1200

Here is the launch file

<launch>

<include file="$(find file_server)/launch/ros_sharp_communication.launch">
    <arg name="port" value="9090" />
</include>

<arg name="base" default="$(env TURTLEBOT_BASE)" />
<arg name="stacks" default="$(env TURTLEBOT_STACKS)" />
<arg name="3d_sensor" default="$(env TURTLEBOT_3D_SENSOR)" />
<arg name="urdf_file" default="$(find xacro)/xacro.py '$(find turtlebot_description)/robots/$(arg base)_$(arg stacks)_$(arg 3d_sensor).urdf.xacro'" />

<param name="robot/name" value="Turtlebot2" />
<param name="robot_description" command="$(arg urdf_file)" />
</launch>

kalepo, 2022-07-13:
<launch>

<include file="$(find file_server)/launch/ros_sharp_communication.launch">
    <arg name="port" value="9090" />
</include>

<arg name="base" default="$(env TURTLEBOT_BASE)" />
<arg name="stacks" default="$(env TURTLEBOT_STACKS)" />
<arg name="3d_sensor" default="$(env TURTLEBOT_3D_SENSOR)" />
<arg name="urdf_file" default="$(find xacro)/xacro.py '$(find turtlebot_description)/robots/$(arg base)_$(arg stacks)_$(arg 3d_sensor).urdf.xacro'" />

<param name="robot/name" value="Turtlebot2" />
<param name="robot_description" command="$(arg urdf_file)" />
</launch>`

MartinBischoff, 2022-07-22:
Please refer to [ROS Answers](https://answers.ros.org/questions/) for how to edit launch files.

closed 2022-07-22 by MartinBischoff