siemens/ros-sharp #436: Unsubscribe and re-subscribe to different joint state topics at runtime

Gescheld, 2022-08-18:
* [X] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stack Overflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [X] I have thoroughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

---
## I have a question! ##

* [X] I searched the [Wiki](https://github.com/siemens/ros-sharp/wiki), [open](https://github.com/siemens/ros-sharp/issues) and [closed](https://github.com/siemens/ros-sharp/issues?q=is%3Aissue+is%3Aclosed) issues for an answer. I tried my best to find the answer by myself without success. I believe that the discussion we will have in this issue, and the solutions we might find, will help me, and likely other community members who have a similar problem.

**Here is my question:**
I want to unsubscribe from my current joint state topic and subscribe to an other joint state topic at runtime with the same robot game object. Is there a way to this at runtime? 

So far I wait unitl I press a button to subscribe to a joint state topic. But I couldn't figure out how to unsubscribe and re-subscribe. I tried to remove/destroy the comoponents and to add them again with the different topic name, but that didn't work either.  Also, I tried to disable the JointStateSubscriber varibale (SetSubscribeJointStates(false)) and enable it again with a different topic at the JointStateSubscriber.cs component.

Those are the scripts components that are added at runtime: 
- [JointStatePatcher.cs](https://github.com/siemens/ros-sharp/blob/8548ed3086212d25faba574d6846bb70cd8881ca/com.siemens.ros-sharp/Assets/RosSharp/Scripts/RosBridgeClient/MessageHandling/JointStatePatcher.cs)
- [JointStateSubscriber.cs](https://github.com/siemens/ros-sharp/blob/8548ed3086212d25faba574d6846bb70cd8881ca/com.siemens.ros-sharp/Assets/RosSharp/Scripts/RosBridgeClient/RosCommuncation/JointStateSubscriber.cs)

Relevant Code:
```
1  RosSharpObject.AddComponent<RosSharp.RosBridgeClient.JointStatePatcher>();
2  // Set Urdf Robot gameobject
3  RosSharpObject.GetComponent<RosSharp.RosBridgeClient.JointStatePatcher>().UrdfRobot = robot_hologram.GetComponent<UrdfRobot>();
4 // Set subscribing value for joint states --> adds JointStateSubscriber.cs to game object
5  RosSharpObject.GetComponent<RosSharp.RosBridgeClient.JointStatePatcher>().SetSubscribeJointStates(true);
6  RosSharpObject.GetComponent<RosSharp.RosBridgeClient.JointStateSubscriber>().Topic = TopicDropdown.options[TopicDropdown.value].text;
```

* My OS is: Windows 10
* My Unity Version is: 2020.3.16f1
* My ROS Distribution is: Noetic
* My Build Target Platform is: UWP

MartinBischoff, 2022-08-24:
Hi @Gescheld
you will have to modify [UnitySubscriber](https://github.com/siemens/ros-sharp/blob/master/com.siemens.ros-sharp/Assets/RosSharp/Scripts/RosBridgeClient/RosCommuncation/UnitySubscriber.cs).
This basic version just subscribes on `Start()`.

You have to add e.g. the following functionality in e.g. `Update()`:
if your `doChangeSubscription` condition is satisfied: unsubscribe and resubscribe

Gescheld, 2022-08-29:
Hi @MartinBischoff,

Thank you for the fast answer. I don't have the opportunity to try this at the moment, but I will test it next time :)