siemens/ros-sharp #441: LaserScanPublisher not giving output

amelia413, 2022-11-17:
* [ ] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stack Overflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [ ] I have thoroughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

---
## I have a question! ##

* [ ] I searched the [Wiki](https://github.com/siemens/ros-sharp/wiki), [open](https://github.com/siemens/ros-sharp/issues) and [closed](https://github.com/siemens/ros-sharp/issues?q=is%3Aissue+is%3Aclosed) issues for an answer. I tried my best to find the answer by myself without success. I believe that the discussion we will have in this issue, and the solutions we might find, will help me, and likely other community members who have a similar problem.

**Here is my question:**
I want to use the lidar data from Unity in ROS. I'm working on the turtlebot3. I have added the LaserScanPublisher in the unity side. Is that enough to read the ranges[] in ROS or do I have to do something else ?

---
## I found a bug! ##
* [ ] I am using the latest ROS# version available here on the master branch.
* [ ] I am adding all required information, code and data files, screenshots and log files so that you can reproduce the problem.

* My OS is: 
* My Unity Version is: 2019.4
* My ROS Distribution is: Noetic 
* My Build Target Platform is: 

**Here is my bug description:**

Perform the following steps reproduce the bug:

Observed results:

Expected results:

Relevant Code:
```
<!-- Insert code here. Optionally attach code or data files to this issue. -->
```
---
## I have a feature request! ##

**I want to do this:**

**For being able to do it, I wish ROS# had the following feature:**

MartinBischoff, 2022-11-17:
Hi @amelia413 as I already wrote [here](https://forum.unity.com/threads/ros-ros-ros-unity-communication-package.508725/#post-8592622), a publisher is enough. Please refer to the publisher references included in ROS# to learn and understand.

MartinBischoff, 2024-01-22:
outdated / closing due to inactivity

closed 2024-01-22 by MartinBischoff