siemens/ros-sharp #444: Received a message without an op.

sstoleriu, 2023-06-19:
* [x ] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stack Overflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [x ] I have thoroughly read [the Contributing Guideline](Contributing.md) and writing this issue is the right thing to do in my case.

---
## I have a question! ##

* [ x] I searched the [Wiki](https://github.com/siemens/ros-sharp/wiki), [open](https://github.com/siemens/ros-sharp/issues) and [closed](https://github.com/siemens/ros-sharp/issues?q=is%3Aissue+is%3Aclosed) issues for an answer. I tried my best to find the answer by myself without success. I believe that the discussion we will have in this issue, and the solutions we might find, will help me, and likely other community members who have a similar problem.

**Here is my question:**
When I want to subscribe to an topic with HoloLens 2 application, I receive the message "[Client x] Received a message without an op." In Play mode works fine, I receive the message from topic.

* My OS is: Windows 10
* My Unity Version is: 2020.3.33f1
* My ROS Distribution is: ros-sharp
* My Build Target Platform is: Microsoft Hololens 2

HaomiaoZ, 2023-06-21:
Try https://github.com/EricVoll/ros-sharp. It looks like UWP needs a different protocol.

sstoleriu, 2023-06-22:
@HaomiaoZ  It works! Thank you!

closed 2023-06-22 by sstoleriu