siemens/ros-sharp #450: Many calls to unsubscribe, subscribe cause Web Socket Net to stop receiving

PetterVMC, 2023-12-20:
Hi!

In my current Unity simulation I need to unsubscribe during runtime based on application logic, and then later re-subscribe to the same topics again. This cycle can repeat many times.

I noticed that after unsubscribing / re-subscribing multiple times, I suddenly stopped receiving any messages at all (even on other topics that I permanently subscribed to). After some debugging, it I noticed that the RosSocket's function Receive() suddently stops being called entirely.

This problem only seems to happen if protocol is "Web Socket Net". The other protocol "Web Socket Sharp" seems to work fine.

Also, my Unity simulation subscribe to about 25 Float32, which are being published from an ubuntu ROS node at 100 Hz. If I lower the data amount or publish frequency the problem is less prone to happen (or does not happen at all).

Anyone having similar experiences, or any insight on if this could be an actual bug in the Web Socket Net protocol?

Unity version: 2022.3.13f1 (Windows 10)
ROS distribution: Melodic (Ubuntu 18.04)

PetterVMC, 2025-12-10:
I will try to test this again with the latest version whenever I have time. In the mean time it's fine to keep this issue closed.

memrecakal, 2025-12-10:
Hi @PetterVMC, I wasn't honestly expecting you to reply after two years 😄 Please share your test results if the issue persists.