siemens/ros-sharp #452: I cannot build the Unity Mixed Reality Project after adding ROS#

tngrbz, 2024-03-12:
Issue Template © Siemens AG, 2017-2018
Author: Dr. Martin Bischoff (martin.bischoff@siemens.com)

**Here is my problem:**

---

* My OS is: 
* My Unity Version is: 
* My ROS Distribution is: 
[EditorLogs.txt](https://github.com/siemens/ros-sharp/files/14571031/EditorLogs.txt)

* My Build Target Platform is: 

Perform the following steps reproduce the bug:

[EditorLogs.txt](https://github.com/siemens/ros-sharp/files/14571035/EditorLogs.txt)

memrecakal, 2024-03-12:
Hi @tngrbz 

Thanks for reaching out. Are you using IL2CPP scripting background in Unity? If so, can you switch to Mono (.NET Framework) and try again?

tngrbz, 2024-03-14:
Hi @memrecakal

Thanks for the reply. However it still fails to build for Android. I get the following error.
[EditorLogs2.txt](https://github.com/siemens/ros-sharp/files/14600238/EditorLogs2.txt)

memrecakal, 2024-03-14:
I see. Where did you download ROS# from? Unity asset store or the Github repo?

tngrbz, 2024-03-14:
From the GitHub repo.

memrecakal, 2024-03-14:
> We suspected this problem and have already downgraded our external dependencies, but the main ros-sharp repo hasn't been updated yet as we are still testing it.

As I said, we are still testing the downgraded version before updating the repo again. The Unity Asses Store version is still using the old .NET version with old external libraries. Can you reinstall ROS# from there and test again? I think that should fix the problem. I myself used that particular version to build an Android app and it worked perfectly.

memrecakal, 2024-03-22:
@tngrbz Did you make any progress? I'd like to close the issue if so.

tngrbz, 2024-03-22:
Hi sorry, I still get an error even after downloading the asset from the unity store. Again I switched to .NET Framework and selected NEWTON_JSON on RosConnector. Here is the error.
[Logs.txt](https://github.com/siemens/ros-sharp/files/14722328/Logs.txt)

memrecakal, 2024-03-22:
It seems that Microsoft.Bcl.AsyncInterfaces.dll is still causing problems. From the log I can see that the 'System.Text.Json.dll' is looking for the 1.0.0 version of Microsoft.Bcl.AsyncInterfaces, which might be deleted in the new Unity 22 version. After all, the ROS# version you are currently using is built for 19 LTS. The easiest thing to do right now would be to manually download the Microsoft.Bcl.AsyncInterfaces.dll 1.0.0, I think, but this might create an ambiguous reference problem within Unity, still worth a try. If that doesn't solve the problem, I would suggest downgrading to Unity 2019 LTS.

tngrbz, 2024-03-22:
Okay, I will give a shot. Since Meta recommends the user to use at least Unity 2022, I will probably look for a different way to use ROS# rather than downgrading. But thanks for the help 👍

memrecakal, 2024-03-30:
Hi @tngrbz, were you able to resolve your issue?

tngrbz, 2024-04-01:
hi @memrecakal, I needed to extract the position and orientation of the headset, I am just doing it using Oculus App and Meta Link, thus I don't have to build the app, I just have to click on play button on Unity. I tested it works just fine like that.

tngrbz, 2024-04-03:
hi @memrecakal, I tried using com.unity.xr.openxr packages which are standard Unity packages, instead of using com.meta.xr.sdk.all, which is the standard Meta package, and it is built fine.

closed 2024-04-14 by memrecakal