siemens/ros-sharp #464: Cannot Transfer URDF From ROS

hijimasa, 2024-11-01:
* [x] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stack Overflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [x] I have thoroughly read [the Contributing Guideline](Contributing) and writing this issue is the right thing to do in my case.

---

* [x] I am using the latest ROS# version available here on the master branch.
* [x] I am adding all required information, code and data files, screenshots and log files so that you can reproduce the problem.

* My OS is: Ubuntu 22.02
* My Unity Version is: Unity 6 (6000.0.25f1)
* My ROS Distribution is: ROS 2 humble
* My Build Target Platform is: Unity Editor

**Here is my bug description:**
I refer to https://github.com/siemens/ros-sharp/wiki/User_App_ROS_TransferURDFFromROS .
Below error occurred when transferring URDF from ROS.
```
FileNotFoundException: Could not find file "/home/hijikata/work/Robot_Unity_App/Assets/Urdf/file_server2/robot_state_publisher:robot_description.urdf"
System.IO.FileStream..ctor (System.String path, System.IO.FileMode mode, System.IO.FileAccess access, System.IO.FileShare share, System.Int32 bufferSize, System.Boolean anonymous, System.IO.FileOptions options) (at <f724beac1c704825ad74571377c52f70>:0)
System.IO.FileStream..ctor (System.String path, System.IO.FileMode mode, System.IO.FileAccess access, System.IO.FileShare share, System.Int32 bufferSize) (at <f724beac1c704825ad74571377c52f70>:0)
(wrapper remoting-invoke-with-check) System.IO.FileStream..ctor(string,System.IO.FileMode,System.IO.FileAccess,System.IO.FileShare,int)
System.Xml.XmlDownloadManager.GetStream (System.Uri uri, System.Net.ICredentials credentials, System.Net.IWebProxy proxy, System.Net.Cache.RequestCachePolicy cachePolicy) (at <342a04f2cf8d45d88209220398fbc274>:0)
System.Xml.XmlUrlResolver.GetEntity (System.Uri absoluteUri, System.String role, System.Type ofObjectToReturn) (at <342a04f2cf8d45d88209220398fbc274>:0)
System.Xml.XmlTextReaderImpl.FinishInitUriString () (at <342a04f2cf8d45d88209220398fbc274>:0)
System.Xml.XmlTextReaderImpl..ctor (System.String uriStr, System.Xml.XmlReaderSettings settings, System.Xml.XmlParserContext context, System.Xml.XmlResolver uriResolver) (at <342a04f2cf8d45d88209220398fbc274>:0)
System.Xml.XmlReaderSettings.CreateReader (System.String inputUri, System.Xml.XmlParserContext inputContext) (at <342a04f2cf8d45d88209220398fbc274>:0)
System.Xml.XmlReader.Create (System.String inputUri, System.Xml.XmlReaderSettings settings, System.Xml.XmlParserContext inputContext) (at <342a04f2cf8d45d88209220398fbc274>:0)
System.Xml.XmlReader.Create (System.String inputUri, System.Xml.XmlReaderSettings settings) (at <342a04f2cf8d45d88209220398fbc274>:0)
System.Xml.Linq.XDocument.Load (System.String uri, System.Xml.Linq.LoadOptions options) (at <ddc5d26c16934c8e8cb0267acccb1a3a>:0)
System.Xml.Linq.XDocument.Load (System.String uri) (at <ddc5d26c16934c8e8cb0267acccb1a3a>:0)
RosSharp.Urdf.Robot..ctor (System.String filename) (at ./Library/PackageCache/com.siemens.ros-sharp/Runtime/Libraries/Urdf/Robot.cs:41)
RosSharp.Urdf.Editor.UrdfRobotExtensions.Create (System.String filename) (at ./Library/PackageCache/com.siemens.ros-sharp/Editor/Urdf/UrdfComponents/UrdfRobotExtensions.cs:44)
RosSharp.RosBridgeClient.TransferFromRosHandler.GenerateModelIfReady () (at ./Library/PackageCache/com.siemens.ros-sharp/Editor/RosBridgeClient/UrdfTransfer/TransferFromRosHandler.cs:122)
RosSharp.RosBridgeClient.TransferFromRosEditorWindow.OnInspectorUpdate () (at ./Library/PackageCache/com.siemens.ros-sharp/Editor/RosBridgeClient/UrdfTransfer/TransferFromRosEditorWindow.cs:141)
UnityEditor.HostView.OnInspectorUpdate () (at /home/bokken/build/output/unity/unity/Editor/Mono/HostView.cs:452)
```
As you can see from the first line, the file name is wrong.
The URDF parameter (robot_state_publisher:robot_description) seems to be related, but changing the URDF parameter to “robot_description” does not resolve the issue.

***Perform the following steps reproduce the bug:***
1. launch file_server with below command
   ```
   ros2 launch file_server2 publish_description_r2d2.launch.py
   ```
2. launch Unity Editor
3. Select [RosBridgeClient] -> [Transfer URDF From ROS...]
4. Set "Robot Name Parameter" "r2d2:file_server2"
5. Set "URDF Parameter" "robot_state_publisher:robot_description"
6. Set "Asset Path" "/home/${USER}/work/Robot_Unity_App/Urdf"
7. Press "Read Robot Description" button
8. Click "Yes" button in pop-up dialogue box
 
***Observed results:***
Screenshot is below.
[image]

***Expected results:***
If the tutorial is followed, a robot model should be generated.

hijimasa, 2024-11-02:
It could be improved by changing line 117 and after in TransferRosHandler.cs as follows.
```
            if (EditorUtility.DisplayDialog(
                "Urdf Assets imported.",
                "Do you want to generate a " + robotName + " GameObject now?",
                "Yes", "No"))
            {
#if ROS2
                string[] urdfNames = urdfParameter.Split(':');
                string urdfName = urdfNames[1];
#else
                string urdfName = urdfParameter;
#endif
                Urdf.Editor.UrdfRobotExtensions.Create(Path.Combine(
                    localDirectory,
                    Path.GetFileNameWithoutExtension(urdfName) + ".urdf"));
            }
```
However, I am not familiar enough with C# to know if this is the correct change.

Dabuss, 2024-11-04:
Hi, I made a PR for this problem here : https://github.com/siemens/ros-sharp/pull/466

hijimasa, 2024-11-04:
Thanks for creating the pull request.

closed 2024-11-04 by hijimasa

memrecakal, 2024-11-04:
Hi @hijimasa, 

Thanks for reaching out and testing ROS# with Unity 6 and Ubuntu 22. No matter what I did, I couldn't reproduce the bug, it works on Windows 11 & Unity 2022.3. Checking the error dialogue, `TransferFromRosHandler` is looking for the `robot_state_publisher:robot_description.urdf`, which is not the intended behaviour. I suspect the reason this isn't a problem on Windows is the different execution of the [Path.GetFileNameWithoutExtension Method](https://learn.microsoft.com/en-us/dotnet/api/system.io.path.getfilenamewithoutextension?view=net-8.0) between Windows and Ubuntu, as `:` is not allowed in the Windows path format. 

The quickest workaround I would suggest (if you haven't already) is to wrap the `urdfParameter` with `CutAfterColon` before passing it to `GetFileNameWithoutExtension`. Not the best practice, but it might be a quick fix.