siemens/ros-sharp #473: Image Subscriber Issue

ShawnZhou997, 2025-01-08:
HI, I was trying to sub a ROS Image topic to my HoloLens2 but something went wrong.
After launching the Realsense ROS Node and Unity, the function `ReceiveMessage()` was called only once.
I've tried new another Image topic but also shows the same.
[image]

Here is my simple Script.

```
using System.Collections;
using System.Collections.Generic;
using UnityEngine;
using RosSharp.RosBridgeClient.MessageTypes.Sensor;

namespace RosSharp.RosBridgeClient
{
    public class CustomImageRawSubscriber : UnitySubscriber<MessageTypes.Sensor.Image>
    {
        public static Texture2D global_texture2d;
        public UnityEngine.UI.Image uiImage;
        bool isInitialized = false;
        protected override void ReceiveMessage(Image img)
        {
            Debug.Log(img.encoding);
            if (!isInitialized)
            {
                global_texture2d = new Texture2D((int)img.width, (int)img.height, TextureFormat.RGB24, false);
                isInitialized = true;
            }
            global_texture2d.LoadRawTextureData(img.data);
            global_texture2d.Apply();
            if(uiImage != null)
            {
                Sprite sprite = Sprite.Create(global_texture2d, new Rect(0, 0, global_texture2d.width, global_texture2d.height), new Vector2(0.5f, 0.5f));
                uiImage.sprite = sprite;
            }
        }

        protected override void Start()
        {
            //Topic = "/camera/color/image_raw";
            Topic = "/test_png";
            base.Start();
        }
    }
}
``` 

I am using:
-  Unity 2022.3.50f1c1
-  Ubuntu 20.04 + ROS Noetic
-  Ros#
-  Realsense D435i

What else can I do?

ShawnZhou997, 2025-01-08:
The

memrecakal, 2025-01-09:
Hi @ShawnZhou997, I cannot reproduce your problem exactly, so this may be an abstract answer. The `ReceiveMessage` function is eventually passed to the internal `Subscriber` class (in `Communicators.cs`) as a callback function. By ROS design, it is expected to return quickly. That's why there is an additional `ProcessMessage` function in `RawImageSubscriber.cs`. I would suggest threading and some kind of buffer for processing messages without blocking. Also, if you can, I would suggest using `CompressedImageSubscriber` as it's faster.

ShawnZhou997, 2025-01-09:
> Hi @ShawnZhou997, I cannot reproduce your problem exactly, so this may be an abstract answer. The `ReceiveMessage` function is eventually passed to the internal `Subscriber` class (in [...]

yeah, I overcame it by **subscribing to the Compressed image topic and assigning it to another global variable**. 
Processing it in `Update()` is acceptable now, possibly related to **bandwidth limitations**.
Thank you.