siemens/ros-sharp #475: Error during build 'Cannot build player while editor is importing assets or compiling scripts'

samcsss, 2025-01-30:
* [✔️ ] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stack Overflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [✔️ ] I have thoroughly read [the Contributing Guideline](Contributing) and writing this issue is the right thing to do in my case.

---

* [✔️ ] I am using the latest ROS# version available here on the master branch.
* [ ✔️] I am adding all required information, code and data files, screenshots and log files so that you can reproduce the problem.

* My OS is: Windows 11
* My Unity Version is: Unity 6 (6000.0.34f1)
* My ROS Distribution is: Foxy
* My Build Target Platform is: Windows

**Here is my bug description:**
The player will not build when the ROS# package is installed.

When building the project, during the build process the build window will display the message "Cannot build player while editor is importing assets or compiling scripts"

The build process fails.

All C# scripts were removed from the project so that I could verify nothing on my end was causing the problem.

I have set the Scripting Backend to Mono, and API Compatibility Level to .NET Framework in the Project Settings
I'm using the ROS# 2.1.0 package

Apologies if I've missed something obvious, bit I'm a bit lost with this one. 
From digging into this issue I've read that the main cause of this error is including the UnityEditor namespace in release code, but I can't find anywhere obvious in the package where this is happening. 
All I can think is that the '#" in the package's folder name is giving Unity a hard time

Perform the following steps reproduce the bug:
- Create a new project in Unity 6.
- Build the project - notice that the build succeeds.
- Import the ROS# package via the git repo (https://github.com/siemens/ros-sharp.git?path=/com.siemens.ros-sharp)  (Package Manager > Install package from git URL)
- Rebuild the project - notice that the build will fail

Observed results:

[image]

Expected results:
An new project with only the ROS# package added should be able to successfully build 

(I know that I may be pushing the limits of what may be supported, but the package works fine with Unity 6 when running in the Editor).

Relevant Code:
relevant code inside the package was not immediately apparent

StepanStulov, 2025-02-18:
This is due to the misconfiguration of the assembly definition at the `com.siemens.ros-sharp/Editor/siemens.ros-sharp.Editor.asmdef` file. It's marked as `Any` platform instead of being `Editor` (see picture).

It's **not conventional / incorrect** to surround Editor-only (the ones having `../Editor/..` in their path) scripts with the `#if UNITY_EDITOR` pre-processor directives. That directive is for injecting fragments of Editor code into Runtime code. It is not for Editor-only scripts. Please do not use this solution and fix the assembly definition file instead. That's the "Unity way".

[image]

StepanStulov, 2025-02-19:
This is due to the misconfiguration of the assembly definition at the [com.siemens.ros-sharp/Editor/siemens.ros-sharp.Editor.asmdef](https://github.com/siemens/ros-sharp/blob/master/com.siemens.ros-sharp/Editor/siemens.ros-sharp.Editor.asmdef) file. It's marked as `Any Platform` instead of being `Editor` (see picture).

Note, that it's **not conventional** to surround Editor-only (the ones having `../Editor/..` in their path) scripts with the `#if UNITY_EDITOR` pre-processor directives. That directive is for injecting fragments of Editor code into Runtime code. It is **not** for Editor-only scripts. Please do not use this solution and fix the assembly definition file instead. That's the "Unity way".

[image]

closed 2025-06-02 by memrecakal