siemens/ros-sharp #488: Listen and send out ROS1 and ROS2 topics at the same time

Carton22, 2025-08-04:
* [x] I am at the right place and my issue is directly related to ROS#. General technical questions I would post e.g. at [ROS Answers](https://answers.ros.org/) or [Stack Overflow](https://stackoverflow.com). For library-specific questions I would look for help in the corresponding library forums.
* [x] I have thoroughly read [the Contributing Guideline](Contributing) and writing this issue is the right thing to do in my case.

---

* [x] I searched the [Wiki](https://github.com/siemens/ros-sharp/wiki), [open](https://github.com/siemens/ros-sharp/issues) and [closed](https://github.com/siemens/ros-sharp/issues?q=is%3Aissue+is%3Aclosed) issues for an answer. I tried my best to find the answer by myself without success. I believe that the discussion we will have in this issue, and the solutions we might find, will help me, and likely other community members who have a similar problem.

**Here is my question:**

Carton22, 2025-08-11:
@memrecakal 

I accidentally found a workaround for this problem.
The trick is straightforward. I add 2 scenes at the same time in my Unity Editor.

[image]

Then in each scene, I put a ros connector.
So in my case, for `ROS_communicate`, I set `ROS1` mode. While for the `Empty` scene, I set `ROS2` mode. Then I make sure they are connected to different ports. Finally, I managed to communicate both `ROS1` and `ROS2` messages at the same time.

But this walkaround only works for me if I play in the editor, or play in the simulator (e.g., [the polyspatial play to device feature ](https://docs.unity3d.com/Packages/com.unity.polyspatial.visionos@2.3/manual/PlayToDevice.html)for Apple Vision Pro development).

I was trying to build 2 scenes at the same time into the same Unity app. Maybe the Scenemanagement API provided by Unity would help, but I did not succeed.

Wanna provide more evidence I found here and please let me know if you have further thoughts.

Thanks!

memrecakal, 2025-08-12:
Hi @Carton22,
Thanks for sharing what you've found. This looks like an interesting side effect of how the editor handles scripts with different scenes. I'm not certain, but I'd guess that the objects/scripts from the other scene may still be valid in memory, at least for that session. Because the Editor behaves loosely with hot-reload and scene merging, it might be "catching" both sets of connectors in memory from "lingering" assemblies. I don't think this unpredictable, half-initialized state is officially supported by Unity. Well, at the end of the day, we sometimes achieve a lot with "unintended" Unity ways. After all, it's a playground in which we work with the available tools. But, to be honest, I'm not sure if I would trust this unpredictable behavior in a production :)

Regarding building two scenes with different assemblies at the same time; it's the same compilation, project (assembly)-wide, not scene-wide, whether it's one or a hundred scenes. I'm not sure if the SceneManagement API can perform assembly switches other than controlling scene loading/unloading. As you've experienced, though, Unity is full of surprises sometimes :)

I'm genuinely curious to see what solution you come up with for this situation. Please do share your findings 🚀